alibaba/AliOS-Things #1419: [Bug Report]: spelling error for modbus part

yong171966, 2022-03-15:
### What happened?

[A bug happened!
运行环境:
固件版本:](https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16145223.0.0.595660b10zzFt5#/Python/docs/zh-CN/haas_extended_api/modbus)

[image]

### Version

rel_3.3.0 (Main)

skylarCai, 2022-03-15:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-03-23:
> ### Contact Details
> _No response_
> 
> ### What happened?
> https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16145223.0.0.595660b10zzFt5#/Python/docs/zh-CN/haas_extended_api/modbus
> 
> [image]
> 
> ### Version
> rel_3.3.0 (Main)
> 
> ### What soultions are you seeing the problem on?
> _No response_
> 
> ### Relevant log output
> _No response_

官网已经解决此问题。

closed 2022-03-24 by skylarCai