alibaba/AliOS-Things #1420: 有没有HAAS100 使用485 接口的案例 ， 最好还有MODBUS主机程序~

frankshen2020, 2022-03-15:
### Contact Details

shenchendky@163.com

### What do you want?

我看硬件有485接口，代码里也有485的相关代码：
```c
/* RS485 Macro definition */
#define HAAS_RS485_UART_IDX  1
#define HAAS_RS485_BD_DFT    19200
#define HAAS_RS485_DFT_CFG   NULL

int32_t haas_rs485_init(uart_dev_t *uart_dev);
int32_t haas_rs485_send(const void *data, uint32_t size, uint32_t timeout);
int32_t haas_rs485_recv(void *data, uint32_t expect_size,
                         uint32_t *recv_size, uint32_t timeout);

int32_t haas_rs485_deinit(uart_dev_t *uart_dev);
#endif
```
但是不知道如何使用这些代码，关于参数的注释不是很多。

haas_rs485_send，haas_rs485_recv 这两个函数还好， 不懂的是  haas_rs485_init(uart_dev_t *uart_dev) 函数的参数怎么设置呢？我在初始化时应该怎么写~

有没有HAAS100 使用485 接口的案例 ， 最好还有MODBUS主机程序~

### Version

master (Default)

skylarCai, 2022-03-16:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

frankshen2020, 2022-03-16:
人工支持

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2022年03月16日 08:00 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [alibaba/AliOS-Things] 有没有HAAS100 使用485 接口的案例 ， 最好还有MODBUS主机程序~ (Issue #1420) |

感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，

您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yong171966, 2022-03-16:
你可以参考下面的接口文档及案例：
https://g.alicdn.com/HaaSAI/PythonDoc/library/_modbus.html?highlight=modbus#module-0
https://g.alicdn.com/HaaSAI/PythonDoc/demos/modbus_demo.html?highlight=modbus

目前都是基于python实现的

yong171966, 2022-03-16:
如果你需要使用C实现，也可以参考这个案例：
https://github.com/alibaba/AliOS-Things/blob/master/components/mbmaster/example/mbmaster_example.c

欢迎提出任何疑问！

qisheng-seu, 2022-03-16:
HaaS100共有3个串口，分别是UART0/1/2，其中UART1是给RS485使用的。具体可以餐考文章，
https://blog.csdn.net/HaaSTech/article/details/119751326?spm=a2cre.b5050543.0.0.777a222abzCcUu

frankshen2020, 2022-03-16:
我在程序中初始化485

      while (1) {
          if(haas_rs485_init(NULL) == 0);
          printf("haas_rs485_init success  \r\n");
        aos_msleep(1000);
    };

但是报错

E:\AliHaas100\WifiDemo\hardware\board\haas100\drivers/rs485.h:14:25: error: unknown type name 'uart_dev_t'
   14 | int32_t haas_rs485_init(uart_dev_t *uart_dev);
      |                         ^~~~~~~~~~
E:\AliHaas100\WifiDemo\hardware\board\haas100\drivers/rs485.h:19:27: error: unknown type name 'uart_dev_t'
   19 | int32_t haas_rs485_deinit(uart_dev_t *uart_dev);
      |                           ^~~~~~~~~~

我找到RS485.c
看到里面引用了#include "aos/hal/uart.h"  这个头文件里面有定义
/*
 * UART dev handle
 */
typedef struct {
    uint8_t        port;   /**< uart port */
    uart_config_t  config; /**< uart config */
    void          *priv;   /**< priv data */
} uart_dev_t;

所以不知道为什么报错 error: unknown type name 'uart_dev_t'

yong171966, 2022-03-16:
确保引用到的"aos/hal/uart.h"在构建系统的include path里。你现在是基于哪个solutions编译的？

frankshen2020, 2022-03-16:
基于linksdk_demo  WIFI设备连接阿里云示例 ，include path 在哪里修改呢？

frankshen2020, 2022-03-16:
我按照(https://gitee.com/alios-things/uart/edit/master/README.md)文档中描述的
## 步骤2 添加组件

案例下载完成后，需要在helloworld_demo组件的package.yaml中添加对组件的依赖：

```yaml

depends:
  - uart: master          # helloworld_demo中引入uart组件

```
## 步骤3 下载组件

在已安装了  的开发环境工具栏中，选择Terminal -> New Terminal启动终端，并且默认工作路径为当前工程的workspace，此时在终端命令行中输入：

```shell

aos install uart

```

上述命令执行成功后，组件源码则被下载到了./components/drivers/peripheral/uart路径中。

## 步骤4 添加示例
uart组件的package.yaml中添加example
```yaml
source_file:
  - "src/*.c"
  - "example/uart_example.c"
```

之后，还是报相同的错误 。

添加组件就是增加了include path  吗？

yong171966, 2022-03-23:
对的，yaml的构建方式就是添加了组件也就添加了相关的头文件。可以搜索钉钉群:HaaS开发者A群，群号：34208347入群交流一下。这么效率会快一些！

closed 2022-03-24 by skylarCai