alibaba/AliOS-Things #1421: 希望提供Haas edu k1板子js轻应用语音播放的示例代码

AnsonCode, 2022-03-16:
### What happened?

运行环境：Haas edu k1

希望提供Haas edu k1板子js轻应用语音播放的示例代码

### Version

rel_3.3.0 (Main)

skylarCai, 2022-03-16:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

qisheng-seu, 2022-03-16:
亲，基于haaseduk1我们打造了多个智能语音的案例，例如
[【C语言/ 玩转TinyML之离线语音唤醒】](https://blog.csdn.net/HaaSTech/article/details/120269599)     

JS轻应用的语音案例，目前还在规划排期中，敬请期待

qisheng-seu, 2022-03-16:
HaaS100上python开发的播报音箱示例可以参考下，代码可复用于HaaS EDU k1
[Python轻应用实现语音播报](https://g.alicdn.com/HaaSAI/PythonDoc/demos/sound_1k.html)

YiluMao, 2022-03-23:
@AnsonCode 请问您的问题有没有被解决，如果已解决，请关闭issue

closed 2022-03-24 by skylarCai