alibaba/AliOS-Things #1423: [Bug Report]: 有问题想咨询下大佬

Ryze-b, 2022-03-17:
### Contact Details

2682394086@qq.com

### What happened?

我在综合了温湿度检测和火焰两个案例开发，然后串口日志中会不停地重复一段代码，不知道哪里出现了问题。而且有时候开机会报错dump end！！！，有时候却不会，这种问题应该怎么更改代码？
运行环境:haas edu k1
固件版本:HaaSPython-HaaSEDUK1-v1.0.2

### Version

master (Default)

### What soultions are you seeing the problem on?

eduk1_demo

Ryze-b, 2022-03-17:
[问题.docx](https://github.com/alibaba/AliOS-Things/files/8282508/default.docx)

skylarCai, 2022-03-17:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-03-17:
> 不停地重复一段代码

您上传的word中红框框起来的日志是Wi-Fi在运行过程中打印出来的日志，不影响系统正常运行，忽略就好了。
另外的dump我们再看看，目前看日志应该不够，可能需要你帮忙提供更多的信息。

Ryze-b, 2022-03-17:
> 
[问题.docx](https://github.com/alibaba/AliOS-Things/files/8283871/default.docx)

jinliangli, 2022-03-17:
首先感谢您反馈这个问题， 根据异常现场初步判断wifi_connect的任务栈被踩踏，具体原因还需要内部压测复现再分析一下。

jinliangli, 2022-03-23:
还在压测复现分析中。

Ryze-b, 2022-03-23:
> 

好的

skylarCai, 2022-03-24:
This issue has not been updated for a while, so we close this issue.  If you have further request, please reopen it or add it.  We will response as soon as possible.

本issue因太久未更新而关闭，如果您还有进一步的问题，请reopen本issue，或者新开一个issue，我们会尽快回复您。

closed 2022-03-24 by skylarCai

jinliangli, 2022-03-25:
已找到解决方案， 压测验证中。

jinliangli, 2022-03-25:
已修复, Pull Request:

https://github.com/alibaba/AliOS-Things/pull/1459