alibaba/AliOS-Things #1425: [Bug Report]: esp32 OSS上传报错

rualdeng, 2022-03-18:
### Contact Details

rualdeng@163.com

### What happened?

A bug happened!
运行环境:
固件版本:自行编译加入了OSS
在宽窄的帮助下，用esp32 4M+8M的板子编译加入了OSS。在上传的时候报错。

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

### Relevant log output

```shell
PutObject fail,code:RequestTimeTooSkewed,message:The difference between the request time and the current time is too large.,requestId:62343B2F9849B434365DD543
```

skylarCai, 2022-03-18:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ZH-21000301, 2022-03-21:
code:RequestTimeTooSkewed，说明是设备端的时间和服务端的时间不一致，请更新下设备上的时间再试下

ZH-21000301, 2022-03-21:
> ### Contact Details
> [rualdeng@163.com](mailto:rualdeng@163.com)
> 
> ### What happened?
> A bug happened! 运行环境: 固件版本:自行编译加入了OSS 在宽窄的帮助下，用esp32 4M+8M的板子编译加入了OSS。在上传的时候报错。
> 
> ### Version
> master (Default)
> 
> ### What soultions are you seeing the problem on?
> py_engine_esp32_demo
> 
> ### Relevant log output
> ```shell
> PutObject fail,code:RequestTimeTooSkewed,message:The difference between the request time and the current time is too large.,requestId:62343B2F9849B434365DD543
> ```
code:RequestTimeTooSkewed，说明是设备端的时间和服务端的时间不一致，请更新下设备上的时间再试下

rualdeng, 2022-03-21:
> code:RequestTimeTooSkewed，说明是设备端的时间和服务端的时间不一致，请更新下设备上的时间再试下

python轻应用 sntp.setTime() 返回0，还是报同样的错。要怎么处理呢？

yong171966, 2022-03-21:
[image]

在上面的函数位置增加红色标注的部分即可。

yong171966, 2022-03-21:
@rualdeng 是否可以帮忙提交一个PR？

rualdeng, 2022-03-21:
> @rualdeng 是否可以帮忙提交一个PR？

好的

YiluMao, 2022-03-23:
@rualdeng 请问您的问题有没有被解决，如果已解决，请关闭issue

closed 2022-03-24 by skylarCai