alibaba/AliOS-Things #1428: haas200串口问题

xyjand, 2022-03-19:
### What do you want?

Haas200扩展针上的串口，3，5，7，9一组，13，15，15，19一组，23，25，27，29一组，这三组哪些串口直接拿来可用的？是属于232？485？，有没有更详细关于串口的资料？

### Version

rel_3.3.0 (Main)

skylarCai, 2022-03-19:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

jinliangli, 2022-03-21:
您好，

这三组串口可以使用的。 HaaS200的串口操作使用AliOS Things的uart驱动子系统， 可以通过uart 驱动子系统提供的API进行操作， https://github.com/alibaba/AliOS-Things/blob/master/components/drivers/peripheral/uart/include/aos/uart.h

HaaS200硬件介绍
https://blog.csdn.net/HaaSTech/article/details/117694610

不属于232、485。属于UART。

jinliangli, 2022-03-21:
HaaS200 串口：
https://blog.csdn.net/qq_20031585/article/details/119847056

YiluMao, 2022-03-23:
请问您的问题有没有被解决，如果已解决，请关闭issue

YiluMao, 2022-03-23:
@xyjand 请问您的问题有没有被解决，如果已解决，请关闭issue

closed 2022-03-23 by xyjand

xyjand, 2022-10-11:
我是：轻应用JS进行开发，针对Haas200,对于软硬件的对应关系app.json，我找不到？你提供的联接里，也没有这个文件。

xyjand
 
发件人： Jinliang Li
发送时间： 2022-03-21 10:25
收件人： alibaba/AliOS-Things
抄送： xyjand; Author
主题： Re: [alibaba/AliOS-Things] haas200串口问题 (Issue #1428)
您好，
这三组串口可以使用的。 HaaS200的串口操作使用AliOS Things的uart驱动子系统， 可以通过uart 驱动子系统提供的API进行操作， https://github.com/alibaba/AliOS-Things/blob/master/components/drivers/peripheral/uart/include/aos/uart.h
HaaS200硬件介绍
https://blog.csdn.net/HaaSTech/article/details/117694610
不属于232、485。属于UART。