alibaba/AliOS-Things #1430: haas200 更新固件

xyjand, 2022-03-21:
### Contact Details

xyjand@yeah.net

### What do you want?

选择haas200，C, 轻应用示例，准备更新主板的固件，在编译阶段就有报错：
	change dir to E:\PROJECT\haas\test\t5_c\solutions\test_demo. 
	Build Solution by haas200 
	scons: Reading SConscript files ...
	There is no depend component `curl` found! 
	========
	[{'curl': 'master ? <!CHIP_HAAS1000>'}]
	1
	Reverse value
	lost component check : need install this component
	Try to install missing component...
	Start to install components... 
	Can't find component curl. 
	No component need to install. 
	There is no depend component `curl` found! 
	========
	[{'curl': 'master ? <!CHIP_HAAS1000>'}]
	1
	Reverse value
	lost component check : need install this component
	Try to install missing component...
	Start to install components... 
	Can't find component curl. 
	No component need to install. 
	Please make sure  [{'curl': 'master ? <!CHIP_HAAS1000>'}] are valid, and try it again. 
	make: *** [all] Error -1
	Build takes 0 Minutes 4 Seconds.

	终端将被任务重用，按任意键关闭。

### Version

rel_3.3.0 (Main)

skylarCai, 2022-03-21:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

jinliangli, 2022-03-21:
您好，

这个问题是components/oss组件依赖curl组件， 但是HaaS Studio没有正确解析依赖自动下载curl组件。 我们会查看解决。 

临时解决方法： 执行命令在components组件下“git clone https://gitee.com/alios-things/curl.git”下载curl到components/curl。

YiluMao, 2022-03-23:
@xyjand 请问您的问题有没有被解决，如果已解决，请关闭issue

closed 2022-03-24 by skylarCai