alibaba/AliOS-Things #1431: 查找haas200的管脚定义

xyjand, 2022-03-21:
### Contact Details

xyjand@yeah.net

### What do you want?

1、通过haas studio，下载的haas200，js轻应用项目，都不带关于haas200主板的示例代码，导致没有资料可以查到针对这块主板的app.json数据。不知道 uart等硬件的定义。请提供可以查询到针对haas200的完整配置app.json 及 针对这个主板的示例代码的连接？在钉钉百事通上查过了，没找到.

### Version

rel_3.3.0 (Main)

skylarCai, 2022-03-21:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

jinliangli, 2022-03-21:
看一下这个HaaS200硬件介绍呢：https://blog.csdn.net/HaaSTech/article/details/117694610

xyjand, 2022-03-21:
你发我的资料，一个是Ｃ语言的，一个是板上硬件资源介绍。难道就没有类似与haas100一样直接了当的json文件吗？（我本来是用haas100开发好了，但这个板子停产了，不利用后续的使用，没办法转到haas200上，我的应用基于wifi+uart ,唉！）
haas100的定义文件如下：
    {
    "io":{
        "D1":{
            "type":"GPIO",
            "port":32,
            "dir":"output",
            "pull":"pulldown"
        },
        "D2":{
            "type":"GPIO",
            "port":33,
            "dir":"output",
            "pull":"pulldown"
        },
        "D3":{
            "type":"GPIO",
            "port":38,
            "dir":"output",
            "pull":"pulldown"
        },
        "D4":{
            "type":"GPIO",
            "port":39,
            "dir":"output",
            "pull":"pulldown"
        },
        "I2C1":{
            "type":"I2C",
            "port":0,
            "addrWidth":7,
            "freq":100000,
            "mode":"master",
            "devAddr":118
        },
        "I2C2":{
            "type":"I2C",
            "port":0,
            "addrWidth":7,
            "freq":100000,
            "mode":"master",
            "devAddr":118
        },
        "ADC0":{
            "type":"ADC",
            "port":0,
            "sampling":12000000
        },
        "ADC1":{
            "type":"ADC",
            "port":1,
            "sampling":12000000
        },
        "ADC2":{
            "type":"ADC",
            "port":2,
            "sampling":12000000
        },
        "PWM0":{
            "type":"PWM",
            "port":0
        },
        "PWM1":{
            "type":"PWM",
            "port":1
        },
        "PWM2":{
            "type":"PWM",
            "port":2
        },
        "PWM3":{
            "type":"PWM",
            "port":3
        },
        "UART2":{
            "type":"UART",
            "port":2,
            "dataWidth":8,
            "baudRate":115200,
            "stopBits":1,
            "flowControl":"disable",
            "parity":"none"
        },
        "SPI0":{
            "type":"SPI",
            "port":0,
            "mode":"master",
            "freq":3250000
        }
    },
    "debugLevel":"DEBUG"
}我对不需要用的移除就是了。这里的ＵＡＲＴ,直接是port:1 或 port:2 ；隐藏了针对管脚的引用。我只会轻应用JS的使用，我不知道你能不能明白我的意思？我是不会Ｃ语言的，也不会修改管脚去对应PORT

xyjand
 
发件人： Jinliang Li
发送时间： 2022-03-21 10:43
收件人： alibaba/AliOS-Things
抄送： xyjand; Author
主题： Re: [alibaba/AliOS-Things] 查找haas200的管脚定义 (Issue #1431)
看一下这个HaaS200硬件介绍呢：https://blog.csdn.net/HaaSTech/article/details/117694610

jinliangli, 2022-03-22:
关于board.json可以参考这个
https://hli.aliyuncs.com/haasai/config/haas200/board.json

这个是uart port和pin脚的关系
static const PinMap CSI_PinMap_UART_TX[] = {
    { PA_12,    UART_3, PIN_DATA(PullUp, PINMUX_FUNCTION_UART), },
    { PA_18,    UART_0, PIN_DATA(PullUp, PINMUX_FUNCTION_UART), },
    { PA_7,     UART_2, PIN_DATA(PullUp, PINMUX_FUNCTION_LOGUART), },
    { NC,       NC,     0, },
};

YiluMao, 2022-03-23:
@xyjand 请问您的问题有没有被解决，如果已解决，请关闭issue

closed 2022-03-23 by xyjand