alibaba/AliOS-Things #1432: haas轻应用→haaspython文档错误

niiutuut, 2022-03-21:
### Contact Details

1483206119@qq.com

### What happened?

HaaS 轻应用→HaaS Python→HaaS轻应用扩展库→Modbus串行通信协议→recv、send文档描述错误
1、recv示例与send示例弄反了
2、send函数原型错误
[image]

[haas python](https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b92276856.0.0.17a360b1tEuLlx#/Python/docs/zh-CN/haas_extended_api/modbus?id=send-%e4%bb%8e%e6%9c%ba%e5%8f%91%e8%a1%8c%e5%93%8d%e5%ba%94%e6%95%b0%e6%8d%ae%e5%88%b0%e4%b8%bb%e6%9c%ba) 

### Version

master (Default)

skylarCai, 2022-03-21:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zy20220224, 2022-03-23:
问题已经收到，我们会尽快订正文档并发布到官网

closed 2022-03-28 by skylarCai