alibaba/AliOS-Things #1435: [Feature Request]: OSS上传时间优化

rualdeng, 2022-03-21:
### Contact Details

rualdeng@163.com

### What do you want?

python轻应用，esp32,wifi上传234K文件，用时5S。能否优化到2S以内，产品Demo开发测试有几个月了，APP已经开发完成。设备需要上传文件速度要快（2S内可接受）。现在卡在最后一个环节了，这个解决就可以打板，小规模测试了。

### Version

master (Default)

skylarCai, 2022-03-21:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

rualdeng, 2022-03-21:
ESP32
同一WIFI环境，暂无法测试多个WIFI环境。
上传文件大小        上传时长
       77K                      2S
      150K                    3-4S
      234K                      5S

现需求是234K文件，上传能在2S内。谢谢！

yong171966, 2022-03-23:
同OSS同学确认了，上述数据在目前的版本上是合理的！

closed 2022-03-28 by skylarCai