alibaba/AliOS-Things #1439: [Bug Report]:

liyunkai, 2022-03-23:
### Contact Details

liyunkai@xiaoantech.com

### What happened?

A bug happened!
运行环境:mac
固件版本:

点 串口 标志 按提示操作   ctrl+T  && p && 2 
之后  terminal会卡死没反应
背景：昨天调试没问题。今天重新连上跑程序发现串口连不上了

### Version

master (Default)

### What soultions are you seeing the problem on?

amp_demo

liyunkai, 2022-03-23:
[image]

skylarCai, 2022-03-23:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zetamind, 2022-03-23:
请先确认是否串口异常或被占用 ？ mac 可以直接使用 picocom 验证，使用命令 picocom -b 115200 /dev/ttyUSB0 （115200为串口波特率，/dev/ttyUSB0 为串口端口），如果任然异常，可以尝试重新插拔串口，reset板子或者重启电脑尝试。

liyunkai, 2022-03-23:
除了重启电脑，重新插拔，Reset板子都试过了。vscode选完会卡死。
现在我是用命令行做的

zetamind, 2022-03-31:
用命令行做是正常的吗 ？ 其实我是想了解用其他的串口工具是否卡死，了解是工具这边的问题还是板子串口已经异常。

closed 2022-04-19 by skylarCai