alibaba/AliOS-Things #1440: [Bug Report]:

liyunkai, 2022-03-23:
### What happened?

A bug happened!
运行环境:CS600S
固件版本:？
跑MQTT示例，网络连接正常(1000毫秒闪灯)，但后续catch，suc，fail都没有跑到(无100毫秒闪灯)。接下来不知道该如何进行下去了

### Version

master (Default)

### What soultions are you seeing the problem on?

amp_demo

### Relevant log output

```shell
var mqtt = require('mqtt');
var network = require('network');
var net = network.openNetWorkClient();

var mqttClient;

function ArrayToString(fileData) {
    var dataString = "";
    for (var i = 0; i < fileData.length; i++) {
        dataString += String.fromCharCode(fileData[i]);
    }

    return dataString;
}

function onConnect() {
    // subscribe
    mqttClient.subscribe({
        topic: '/x3_prototype',
        success: function () {
            console.log('subscribe [/x3_prototype] success');
        }
    });

    // publish
    setInterval(function () {
        mqttClient.publish({
            topic: '/x3_prototype',
            message: 'this is AMP mqtt test',
            success: function () {
                console.log('publish [/x3_prototype] success');
            }
        });
    }, 2 * 1000);
}

function mqtt_test() {

 var gpio = require('gpio');

 var led = gpio.open({
	 id: 'USER_LED'
 });
 
 setInterval(function () {
	led.toggle();
 }, 1000);

try{
    // create mqtt client
    mqttClient = mqtt.createClient({
        host: 'api.easylink.io',
        port: 1983,
        // username: 'aiot',
        // password: '123',
        success: function () {
            console.log('mqtt connected');

    setInterval(function () {
        led.toggle();
     }, 100);
            
        },
        fail: function () {
            console.log('mqtt connect failed');
            
    setInterval(function () {
        led.toggle();
     }, 100);
           
        }
    });

    mqttClient.on('connect', function () {
        console.log('mqtt connected');
        onConnect();
    });

    mqttClient.on('message', function (topic, payload) {
        console.log('[' + topic + ']  message: ' + ArrayToString(payload));
    });

    
}catch(err){
    setInterval(function () {
        led.toggle();
     }, 100);
}

}

var netStu = net.getStatus();

if (netStu == 'connect') {
    mqtt_test();
} else {
    net.on('connect', function () {
        mqtt_test();
    })
}
```

skylarCai, 2022-03-23:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

liyunkai, 2022-03-23:
MQTT服务收发测试都是正常的。板子连不上

qisheng-seu, 2022-03-23:
您好，HaaS-JS的mqtt组件是支持的，具体使用方法可以参考《[HaaS-JS MQTT组件JSAPI介绍](https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cre.b5050543.0.0.777a222akvBAze#/JavaScript/docs/zh-CN/standard_api/MQTT-%E6%B6%88%E6%81%AF%E9%98%9F%E5%88%97%E9%81%A5%E6%B5%8B%E4%BC%A0%E8%BE%93%E5%8D%8F%E8%AE%AE%E6%A8%A1%E5%9D%97)》

另外，您提到的mqtt connect success/ fail在你的代码中也有看到，例如 console.log('mqtt connected');

closed 2022-03-28 by skylarCai