alibaba/AliOS-Things #1441: [Feature Request]: 怎么控制led灯

wenkaitiantian, 2022-03-23:
### What do you want?

haas edu 教育版怎么控制led灯的开关

### Version

master (Default)

skylarCai, 2022-03-23:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-03-23:
> ### Contact Details
> _No response_
> 
> ### What do you want?
> haas edu 教育版怎么控制led灯的开关
> 
> ### Version
> master (Default)
您好，HaaS EDU K1上面的RGB LED分别接到HaaS1000的GPIO36, GPIO35, GPIO34三根GPIO上([链接](https://blog.csdn.net/HaaSTech/article/details/113985910?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522164801796416781685370979%2522%252C%2522scm%2522%253A%252220140713.130102334.pc%255Fall.%2522%257D&request_id=164801796416781685370979&biz_id=0&utm_medium=distribute.pc_search_result.none-task-blog-2~all~first_rank_ecpm_v1~rank_v31_ecpm-6-113985910.142^v3^pc_search_result_control_group,143^v4^register&utm_term=haas%E6%8A%80%E6%9C%AF%E7%A4%BE%E5%8C%BA&spm=1018.2226.3001.4187))，所以可以通过控制GPIO的高低来控制LED的量灭。
下面是Python软件的代码：

from driver import GPIO

led_r =GPIO()
led_r.open("led_r")
led_r.write(1) # 点亮LED
led_r.write(0) # 熄灭LED

上面代码中open函数中带的"led_r"是设备节点的信息，定义在board.json中，haaseduk1默认的board.json中位于开发板Python固件的/data/lib/board.json中。截取部分board.json的定义如下：

      "led_r": {
        "type": "GPIO",
        "port": 36,
        "dir": "output",
        "pull": "pullup"
      },
      "led_g": {
        "type": "GPIO",
        "port": 35,
        "dir": "output",
        "pull": "pullup"
      },
      "led_b": {
        "type": "GPIO",
        "port": 34,
        "dir": "output",
        "pull": "pullup"
      },

wenkaitiantian, 2022-03-23:
这个解决不了我的问题，GPIO_P07的电压不够，需要借助3v3供电

wenkaitiantian, 2022-03-23:
[image]
我看这个图片的链接可以解决我的问题，但是我不知道需要什么模板

ethanlcz, 2022-03-23:
> 这个解决不了我的问题，GPIO_P07的电压不够，需要借助3v3供电

所以你要kong'z

> 这个解决不了我的问题，GPIO_P07的电压不够，需要借助3v3供电

所以你的场景是通过HaaS EDU K1 扩展口外接了一个LED，外接的LED需要比较大的电流，所以你外面的LED电路加了上拉电路来确保足够电流来驱动LED，对吧？
如果是这样的话，同样可以参考led_r的案例，需要如下两步：
1. 在[board.json](https://gitee.com/haasedu/haasedu/tree/master/3-%E6%99%BA%E6%85%A7%E5%86%9C%E4%B8%9A/%E5%9C%BA%E6%99%AF1-%E8%8A%B1%E5%8D%89%E5%85%BB%E6%A4%8D/1-%E6%B8%A9%E6%B9%BF%E5%BA%A6%E6%A3%80%E6%B5%8B/%E4%BB%A3%E7%A0%81/HaaS_EDU_K1/Python)的基础上，增加一个名为“extLed”的节点，如果其中的port为GPIO的端口号，端口好参考[链接](https://blog.csdn.net/HaaSTech/article/details/113985910?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522164801796416781685370979%2522%252C%2522scm%2522%253A%252220140713.130102334.pc%255Fall.%2522%257D&request_id=164801796416781685370979&biz_id=0&utm_medium=distribute.pc_search_result.none-task-blog-2~all~first_rank_ecpm_v1~rank_v31_ecpm-6-113985910.142%5Ev3%5Epc_search_result_control_group,143%5Ev4%5Eregister&utm_term=haas%E6%8A%80%E6%9C%AF%E7%A4%BE%E5%8C%BA&spm=1018.2226.3001.4187)中“扩展接口”小节的“GPIO在代码中的对应关系”图片中的说明。
"extLed": {
    "type": "GPIO",
    "port": 7,
    "dir": "output",
    "pull": "pullup"
  }
2. 此时对应的python代码可以参考如下：
from driver import GPIO

led =GPIO()
led.open("extLed")
# 点亮或熄灭LED，write参数传入0或者1取决于外围电路的设计
led.write(1) 
led.write(0)

wenkaitiantian, 2022-03-25:
链接led灯，电压不够怎么解决
我链接一个是亮的，链接10个就不亮了

ethanlcz, 2022-04-01:
> 链接led灯，电压不够怎么解决 我链接一个是亮的，链接10个就不亮了

请问下这个问题解决掉了没？

closed 2022-04-08 by skylarCai