alibaba/AliOS-Things #1444: [Bug Report]:

krisaaaaa, 2022-03-23:
### What happened?

A bug happened!
运行环境:win10 VScode
固件版本:haas edu k1 python轻应用

### Version

master (Default)

### What soultions are you seeing the problem on?

eduk1_demo, haas_dev_demo

### Relevant log output

```shell
请问我创建项目时选择了串口COM3，但在部署运行时这样是什么原因呢(terminal)，温湿度上云haaseduk1。

Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Users\Admin\workspace\HaaSPython> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\Admin\workspace\HaaSPython\.utility\python\transymodem.py -d COM1                 -b 1500000 c:\Users\Admin\workspace\HaaSPython\upgrade\pyamp.zip
Namespace(baudrate='1500000', device='COM1', files=['c:\\Users\\Admin\\workspace\\HaaSPython\\upgrade\\pyamp.zip'])
the filepath is abs path
serial port is COM1
transfer baudrate is 1500000
filepath is c:\Users\Admin\workspace\HaaSPython\upgrade\pyamp.zip
Traceback (most recent call last):
  File "c:\Users\Admin\workspace\HaaSPython\.utility\python\transymodem.py", line 72, in download_file
    serialport.open()
  File "C:\ProgramData\aos\miniconda3\lib\site-packages\serial\serialwin32.py", line 80, in open
    self._reconfigure_port()
  File "C:\ProgramData\aos\miniconda3\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'The parameter is incorrect.', None, 87)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\Admin\workspace\HaaSPython\.utility\python\transymodem.py", line 247, in <module>
    main2()
  File "c:\Users\Admin\workspace\HaaSPython\.utility\python\transymodem.py", line 242, in main2
    download_file(downloadconfig["serialport"],
  File "c:\Users\Admin\workspace\HaaSPython\.utility\python\transymodem.py", line 74, in download_file
    raise Exception("Failed to open serial port: %s!" % portnum)
Exception: Failed to open serial port: COM1!
PS C:\Users\Admin\workspace\HaaSPython>
```

skylarCai, 2022-03-23:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

senwangal, 2022-03-23:
这行打印的意思串口没有成功打开，有可能是重新开机导致串口号变了，可以通过点击底部串口按钮
[image]
手动指定一下实际的串口看看

senwangal, 2022-03-23:
然后再推送看看

senwangal, 2022-03-28:
从这份log看是正常的，这两个wifi _config 用于提供wifi重连服务打印信息可以忽略。

senwangal, 2022-03-28:
> 
这里可以认为串口已经找到了，可以继续推送操作（建议按键ctrl+]退出串口）

closed 2022-03-31 by skylarCai