alibaba/AliOS-Things #1445: haas轻应用文档错误

niiutuut, 2022-03-24:
### What happened?

haas轻应用 - Driver - 系统外设库 - PWM - 脉宽调制功能接口 - open - 打开PWM
文档表格格式错误
[image]

### Version

master (Default)

skylarCai, 2022-03-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-03-24:
> ### Contact Details
> _No response_
> 
> ### What happened?
> haas轻应用 - Driver - 系统外设库 - PWM - 脉宽调制功能接口 - open - 打开PWM 文档表格格式错误 [image]
> 
> ### Version
> master (Default)
> 
> ### What soultions are you seeing the problem on?
> _No response_
> 
> ### Relevant log output
> _No response_

谢谢帮忙指正错误，我们预计今天把这个问题修复掉并上线官网。

ethanlcz, 2022-03-24:
> ### Contact Details
> _No response_
> 
> ### What happened?
> haas轻应用 - Driver - 系统外设库 - PWM - 脉宽调制功能接口 - open - 打开PWM 文档表格格式错误 [image]
> 
> ### Version
> master (Default)
> 
> ### What soultions are you seeing the problem on?
> _No response_
> 
> ### Relevant log output
> _No response_

您好，该问题已经修复。

ethanlcz, 2022-03-24:
[image]

closed 2022-03-31 by skylarCai