alibaba/AliOS-Things #1452: [Feature Request]:需要一份有关recorder API的完整文档

rualdeng, 2022-03-24:
### Contact Details

rualdeng@163.com

### What do you want?

[feature request](https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/audio)只有audio,没有recorder，需要一份完整的有关recorder,API的文档。谢谢

### Version

master (Default)

rualdeng, 2022-03-24:
https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/audio

skylarCai, 2022-03-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-03-24:
> ### Contact Details
> [rualdeng@163.com](mailto:rualdeng@163.com)
> 
> ### What do you want?
> (https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/audio) 只有audio,没有recorder，需要一份完整的有关recorder,API的文档。可用在m5stack core2板上的。谢谢！
> 
> ### Version
> master (Default)

目前暂未支持录音功能，请问您是有什么场景下的具体需求吗？

rualdeng, 2022-03-24:
> 目前暂未支持录音功能，请问您是有什么场景下的具体需求吗？

我们现在做的产品需要录音功能，已经和python轻应用团队对接很久了。之前一直在做4G产品，那边有个小技术升级。WIFI版本我们选择以m5stack core2为开发板。基本的python业务逻辑已经做好了，APP也做好了。
还有需要SD挂载的功能。

closed 2022-04-19 by skylarCai