alibaba/AliOS-Things #1453: [Feature Request]: 如何获取esp32的设备号

rualdeng, 2022-03-24:
### Contact Details

rualdeng@163.com

### What do you want?

板：m5stack core2
如何获取esp32的设备号，以用来做动态设备注册上阿里云。

### Version

master (Default)

skylarCai, 2022-03-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

sunlrain, 2022-03-24:
@rualdeng - 目前推荐的方式是三元组的方式入网， 直接配置好设备信息就可以连接到物联网平台。你是指云端是一型一密下面的设备动态注册吗？

rualdeng, 2022-03-24:
> @rualdeng - 目前推荐的方式是三元组的方式入网， 直接配置好设备信息就可以连接到物联网平台。 看你的问题，是已经在云端创建好了一型一密的产品，然后再通过设备动态注册吗？

是的，如果用三元组方式入网，需要蓝牙方式写入设备，这块又需要我们额外开发。增加时间成本。如果获取设备号，我们提前在云端创建好产品和设备，动态注册更符合我们的需求。之前4G产品也是用的动态注册。

sunlrain, 2022-04-07:
> > @rualdeng - 目前推荐的方式是三元组的方式入网， 直接配置好设备信息就可以连接到物联网平台。 看你的问题，是已经在云端创建好了一型一密的产品，然后再通过设备动态注册吗？
> 
> 是的，如果用三元组方式入网，需要蓝牙方式写入设备，这块又需要我们额外开发。增加时间成本。如果获取设备号，我们提前在云端创建好产品和设备，动态注册更符合我们的需求。之前4G产品也是用的动态注册。

但是M5stack是Wi-Fi联网的，需要配置你所在环境中的Wi-Fi/密码来上网，跟4G的插SIM卡使用不太一样。Wi-Fi热点和密码你是准备怎么配置呢？

closed 2022-04-19 by skylarCai