alibaba/AliOS-Things #1454: [Bug Report]: m5stack core2 播放mp3报错

rualdeng, 2022-03-24:
### Contact Details

rualdeng@

### What happened?

A bug happened!
运行环境:
固件版本:HaaSPython-ESP32-8M-v1.0.3

### Version

master (Default)

### What soultions are you seeing the problem on?

helloworld_demo

### Relevant log output

```shell
I (9104) ESP_AUDIO_TASK: media_ctrl_task running...,0x3f802394

----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
|                 ESP_AUDIO-v1.7.0-31-g5b8f999-3072767-09be8fe               |
|                     Compile date: Oct 14 2021-11:00:34                     |
------------------------------------------------------------------------------
I (9144) MP3_DECODER: MP3 init
I (9164) ESP_AUDIO_CTRL: Enter play procedure, src:0
I (9164) ESP_AUDIO_CTRL: Play procedure, URL is ok, src:0
W (9164) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
I (9174) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3fff2304
I (9184) ESP_AUDIO_TASK: It's a decoder
I (9184) ESP_AUDIO_TASK: 1.CUR IN:[IN_file],CODEC:[NULL],RESAMPLE:[44100],OUT:[OUT_iis],rate:0,ch:0,pos:0
W (9194) ESP_AUDIO_TASK: Try to got music format by in stream
I (9204) ESP_AUDIO_TASK: Get codec type occur with element status [0xc], line 197
I (9214) ESP_AUDIO_TASK: 2.Handles,IN:0x3f9138ec,CODEC:0x0,FILTER:0x3f914684,OUT:0x3f9144cc
E (9224) ESP_AUDIO_TASK: NULL does not supported
I (9224) ESP_AUDIO_CTRL: Exit play procedure, ret:81003
the ret is 528387
I (9234) ESP_AUDIO_CTRL: Enter media_ctrl_stop procedure, src:0, type:0, st:ERROR, phase:0
W (9244) ESP_AUDIO_CTRL: [media_ctrl_stop]-Already stopped, status is ERROR, phase:0
I (9254) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:5, Available:5, que:0x3fff2304
I (9254) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_EXIT
I (9264) ESP_AUDIO_TASK: media_ctrl_task has been destroyed
E (9264) AUDIO_ELEMENT: [IN_file] Element already stopped
I (9274) CODEC_ELEMENT_HELPER: The element is 0x3f9142e0. The reserve data 2 is 0x0.
```

skylarCai, 2022-03-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yong171966, 2022-03-24:
请提供详细的复现步骤，包括python命令。

rualdeng, 2022-03-25:
> 请提供详细的复现步骤，包括python命令。

def power():
        axp = Axp192()
        axp.powerAll()
        axp.setSpkEnable(1)

def music():
        print("i am in the music")
        player = Player()
        ret = player.open()
        ret = player.play(uri='file://35.mp3', sync=True)
        print("the ret is",ret)
        player.stop()
        player.close()
    
if  __name__ == '__main__':
    sta_if = network.WLAN(network.STA_IF)
    sta_if.active(True)
    sta_if.connect('xxxx', 'xxxxx')  //此处填写的WIFI帐号和密码
    utime.sleep_ms(500)
    begin = utime.time()
    while not sta_if.isconnected():
        print("the net is none")
        utime.sleep_ms(500)
        print('this time is')
        print(utime.localtime())

    if sta_if.isconnected():
        print(str(sta_if.isconnected()))
        print("联上网了")
        power()
        utime.sleep(1)
        music()

yong171966, 2022-03-31:
uri需要指定到本地的文件或者网络文件，请问你35.mp文件放置的具体系统是什么？

[image]

按照上面的示例，实际上播放的是 lib/audio/下的test.mp3文件

yong171966, 2022-03-31:
具体使用方式请参考如下链接：
https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16145223.0.0.595660b1L36Sdd#/Python/docs/zh-CN/haas_extended_api/audio

closed 2022-04-08 by skylarCai