alibaba/AliOS-Things #1456: [Feature Request]: 想要一份dht11传感器的python文件

Ryze-b, 2022-03-24:
### Contact Details

2682394086@qq.com

### What do you want?

[image]
或者有什么参考资料吗？

### Version

master (Default)

skylarCai, 2022-03-24:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-03-24:
> ### Contact Details
> [2682394086@qq.com](mailto:2682394086@qq.com)
> 
> ### What do you want?
> [image] 或者有什么参考资料吗？
> 
> ### Version
> master (Default)

下个版本（预计4月初）可以将DHT11的驱动开源出去。

Ryze-b, 2022-03-24:
好的，谢谢！

closed 2022-03-31 by skylarCai