alibaba/AliOS-Things #1460: M5需要recorder和SD卡挂载功能，怎么推进这件事？> > ### Contact Details

rualdeng, 2022-03-28:
> > ### Contact Details
> > [rualdeng@163.com](mailto:rualdeng@163.com)
> > ### What do you want?
> > (https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/haas_extended_api/audio) 只有audio,没有recorder，需要一份完整的有关recorder,API的文档。可用在m5stack core2板上的。谢谢！
> > ### Version
> > master (Default)
> 
> 目前暂未支持录音功能，请问您是有什么场景下的具体需求吗？

我们现在做的产品需要录音功能，已经和python轻应用团队对接很久了。之前一直在做4G产品，那边有个小技术升级。WIFI版本我们选择以m5stack core2为开发板。基本的python业务逻辑已经做好了，APP也做好了。
还有需要SD挂载的功能。

_Originally posted by @rualdeng in https://github.com/alibaba/AliOS-Things/issues/1452#issuecomment-1077532684_

skylarCai, 2022-03-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zy20220224, 2022-03-28:
m5stack已经支持sd卡了，你在sd卡这里有特别的需求吗？

rualdeng, 2022-03-29:
> m5stack已经支持sd卡了，你在sd卡这里有特别的需求吗？

需要把文件写入SD卡，昨天查资料弄了一晚，还是挂不上。
axp = Axp192()
axp.setLDOVoltage(2,3300)
axp.setLDOEnable(2,True)
sdcard = SDCard(slot=2, sck=Pin(18), miso=Pin(38), mosi=Pin(23), cs=Pin(4), freq=40000000)
uos.mount(sdcard, '/sd')
print(uos.listdir('/sd'))

有挂载的案例么？

rualdeng, 2022-03-30:
> m5stack已经支持sd卡了，你在sd卡这里有特别的需求吗？

麻烦提供一下python轻应用SD卡挂载案例。没有特别的要求，需要挂载SD和文件系统，写入文件。SDIO的协议版本来支持SD卡的容量，这个SPI以什么方式来支持SD的最大容量，这个板子最大能支持多大容量的卡，能否自行调节支持128G？

zy20220224, 2022-03-31:
暂时没有提供sd卡挂在的接口，后续版本中会支持起来

rualdeng, 2022-04-06:
> 

能否指导怎么弄，我可以自行修改C程序，自己编译。你们的后续版本比较慢，时间上来不急。

yong171966, 2022-04-25:
Recorder已经支持并开源。

closed 2022-04-25 by skylarCai