alibaba/AliOS-Things #1461: [Feature Request]: 卡编译 ,具体需要什么系统环境 python版本等环境问题

kijlove, 2022-03-28:
### Contact Details

kijlee@163.com

### What do you want?

scons: warning: you do not seem to have the pywin32 extensions installed;
        parallel (-j) builds may not work reliably with open Python files.
File "C:\ProgramData\aos\miniconda3\Scripts\scons", line 204, in <module>
scons: Building targets ...
CC out\linksdk\core\aiot_http_api.o
CC out\linksdk\core\aiot_mqtt_api.o
CC out\linksdk\core\aiot_state_api.o
CC out\linksdk\core\sysdep\core_sysdep.o
CC out\linksdk\core\utils\core_auth.o
CC out\linksdk\core\utils\core_diag.o
CC out\linksdk\core\utils\core_global.o
scons: *** [out\linksdk\core\aiot_http_api.o] ImportError : DLL load failed while importing win32file: 找不到指定的程序。
CC out\linksdk\core\utils\core_http.o
Traceback (most recent call last):
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Taskmaster.py", line 255, in execute
    self.targets[0].build()
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Node\__init__.py", line 766, in build        
    self.get_executor()(self, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 397, in __call__
    return _do_execute_map[self._do_execute](self, target, kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 128, in execute_action_list
    status = act(*args, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 1052, in __call__
    return act(target, source, env, exitstatfunc, presub,
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 1120, in __call__
    return c.__call__(self, target, source, env, *args, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 708, in __call__
    stat = self.execute(target, source, env, executor=executor)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 952, in execute
    result = spawn(shell, escape, cmd_line[0], cmd_line, ENV)
  File "C:\Users\kij13\AppData\Roaming\Python\Python38\site-packages\aostools\make.py", line 128, in win_longcmd_spawn        
    import win32file
ImportError: DLL load failed while importing win32file: 找不到指定的程序。
scons: *** [out\linksdk\core\aiot_mqtt_api.o] ImportError : DLL load failed while importing win32file: 找不到指定的程序。     
Traceback (most recent call last):
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Taskmaster.py", line 255, in execute
    self.targets[0].build()
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Node\__init__.py", line 766, in build        
    self.get_executor()(self, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 397, in __call__
    return _do_execute_map[self._do_execute](self, target, kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 128, in execute_action_list
    status = act(*args, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 1052, in __call__
    return act(target, source, env, exitstatfunc, presub,
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 1120, in __call__
    return c.__call__(self, target, source, env, *args, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 708, in __call__
    stat = self.execute(target, source, env, executor=executor)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 952, in execute
    result = spawn(shell, escape, cmd_line[0], cmd_line, ENV)
  File "C:\Users\kij13\AppData\Roaming\Python\Python38\site-packages\aostools\make.py", line 128, in win_longcmd_spawn        
    import win32file
ImportError: DLL load failed while importing win32file: 找不到指定的程序。
scons: *** [out\linksdk\core\aiot_state_api.o] ImportError : DLL load failed while importing win32file: 找不到指定的程序。    
Traceback (most recent call last):
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Taskmaster.py", line 255, in execute
    self.targets[0].build()
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Node\__init__.py", line 766, in build        
    self.get_executor()(self, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 397, in __call__
    return _do_execute_map[self._do_execute](self, target, kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 128, in execute_action_list
    status = act(*args, **kw)
[... 98 lines cut ...]
    self.targets[0].build()
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Node\__init__.py", line 766, in build        
    self.get_executor()(self, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 397, in __call__
    return _do_execute_map[self._do_execute](self, target, kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Executor.py", line 128, in execute_action_list
    status = act(*args, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 1052, in __call__
    return act(target, source, env, exitstatfunc, presub,
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 1120, in __call__
    return c.__call__(self, target, source, env, *args, **kw)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 708, in __call__
    stat = self.execute(target, source, env, executor=executor)
  File "c:\users\kij13\appdata\roaming\python\python38\site-packages\scons\SCons\Action.py", line 952, in execute
    result = spawn(shell, escape, cmd_line[0], cmd_line, ENV)
  File "C:\Users\kij13\AppData\Roaming\Python\Python38\site-packages\aostools\make.py", line 128, in win_longcmd_spawn        
    import win32file
ImportError: DLL load failed while importing win32file: 找不到指定的程序。
scons: building terminated because of errors.
make: *** [all] Error 2
Build takes 0 Minutes 4 Seconds.

### Version

rel_3.3.0 (Main)

skylarCai, 2022-03-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zetamind, 2022-03-28:
看起来像是环境相关问题，windows系统是使用haas studio直接编译的吗？ 还是使用命令行进行编译的 ？
haas studio 会安装相关编译环境，相关环境安装在 miniconda 环境中。但是看起来调用的 python 并不是 miniconda 独立环境中的。另外就是编译需要用到 python3 的版本。

kijlove, 2022-03-28:
直接用的haas studio下面的编译按钮,电脑中有python环境 是3.8  ,未来会有pycharm插件和clion插件吗

zetamind, 2022-03-28:
pycharm 和 clion 插件暂时还没有，直接用插件编译的话，你换一个工程编译也是一样的问题吗 ？ 你有修改过插件路径吗 ？另外vscode 右下角有通知要安装工具的消息吗 ？

kijlove, 2022-03-28:
我下载个ubuntu试试

kijlove, 2022-03-28:
UnboundLocalError: local variable 'tar_path' referenced before assignment:
  File "/home/alios/code/AliOS-Things/solutions/amp_demo/SConstruct", line 11:
    defconfig.build_components()
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/aostools/make.py", line 119:
    self.build_component(component)
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/aostools/make.py", line 103:
    SCons.SConscript(file_name, exports={"env": self.build_env.env.Clone(
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/scons/SCons/Script/SConscript.py", line 660:
    return method(*args, **kw)
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/scons/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/scons/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/alios/code/AliOS-Things/components/cli/SConscript", line 2:
    defconfig.library_yaml()
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/aostools/make.py", line 91:
    self.build_env.build_component(component)
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/aostools/builder.py", line 339:
    env = self.clone_component(component)
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/aostools/builder.py", line 311:
    path = tool.check_toolchain(self.PREFIX, self.PREFIX_PATH)
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/aostools/toolchain.py", line 128:
    self.download(arch)
  File "/home/alios/.aliot/miniconda3/envs/_aos_env/lib/python3.8/site-packages/aostools/toolchain.py", line 114:
    put_string(tar_path)
make: *** [Makefile:12: all] Error 2

kijlove, 2022-03-28:
新换的ubuntu环境 干干净净的 vscode还是不行

kijlove, 2022-03-28:
--- Available ports:
---  1: /dev/ttyS0           'ttyS0'
---  2: /dev/ttyUSB0         'CP2102N USB to UART Bridge Controller'
--- Enter port index or full name: 2
check_uart open serialport: /dev/ttyUSB0 error 

--- Available ports:
---  1: /dev/ttyS0           'ttyS0'
---  2: /dev/ttyUSB0         'CP2102N USB to UART Bridge Controller'

zetamind, 2022-04-20:
aos tools 版本已经更新，1.1.38 版本修复 Ubuntu 异常报错

closed 2022-04-25 by skylarCai