alibaba/AliOS-Things #1462: [Bug Report]: HAAS EDU K1下载不了

rabbit010, 2022-03-28:
### Contact Details

550988594@qq.com

### What happened?

A bug happened!
运行环境:win10
固件版本:刚买的板子

### Version

master (Default)

### What soultions are you seeing the problem on?

helloworld_demo

### Relevant log output

```shell
> Executing task: aos burn <

change dir to D:\job\workspace\solutions\hello_world. 
binary file is ['D:\\job\\workspace\\solutions\\hello_world\\out\\hello_world@haaseduk1.bin']
serial port is COM3
the settings were restored in the file D:\job\workspace\solutions\hello_world\.config_burn
bin_files is  [('D:\\job\\workspace\\solutions\\hello_world\\out\\hello_world@haaseduk1.bin', '0')]
[ScriptPrint] Try to reboot...
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
[ScriptPrint] reboot fail
[ScriptPrint] Please connect the serial port of the board to the PC, then reset the board
```

skylarCai, 2022-03-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yanxiaoyong-ch, 2022-03-28:
从log看你的串口没有任何输出，请检查串口是否正确，检测串口正确方式：连接该串口，进行设备复位，看串口是否有打印输出；如果有输出，可进行烧录

rabbit010, 2022-04-01:
我换了台电脑 还是不能烧录，可能板子里的估计版本有问题？ 问下 怎么重新下载烧录固件里的初始化版本？ 我重新刷机看看？

rabbit010, 2022-04-01:
插入设备 电脑端 可以识别串口cp210 usb转uart 口

rabbit010, 2022-04-01:
检测串口正确方式：连接该串口，进行设备复位，看串口是否有打印输出；如果有输出，可进行烧录// 请截图 让我看行正常打印是什么，谢谢！@[yanxiaoyong-ch](https://github.com/yanxiaoyong-ch)

rabbit010, 2022-04-01:
> Executing task: aos burn <

change dir to C:\Users\55098\workspace\solutions\hello_world. 
binary file is ['C:\\Users\\55098\\workspace\\solutions\\hello_world\\out\\hello_world@haaseduk1.bin']
serial port is COM3
the settings were restored in the file C:\Users\55098\workspace\solutions\hello_world\.config_burn
bin_files is  [('C:\\Users\\55098\\workspace\\solutions\\hello_world\\out\\hello_world@haaseduk1.bin', '0')]
[ScriptPrint] Try to reboot...
[ScriptPrint] change to CLI mode
[ScriptPrint] change to CLI mode
Traceback (most recent call last):
  File "C:\Users\55098\workspace\hardware\chip\haas1000\release/aos_burn_tool/flash_program.py", line 130, in <module>
    main()
  File "C:\Users\55098\workspace\hardware\chip\haas1000\release/aos_burn_tool/flash_program.py", line 125, in main
    burn_bin_files(myconfig["serialport"], myconfig['baudrate'], bin_files)
  File "C:\Users\55098\workspace\hardware\chip\haas1000\release\aos_burn_tool\flash_program_ll
.py", line 123, in burn_bin_files
    match = send_and_match(serialport, b'\n', b'aos boot', 2)
  File "C:\Users\55098\workspace\hardware\chip\haas1000\release\aos_burn_tool\flash_program_ll.py", line 46, in send_and_match
    line = serialport.read_until(b'\n')
  File "C:\ProgramData\aos\miniconda3\lib\site-packages\serial\serialutil.py", line 663, in read_until
    c = self.read(1)
  File "C:\ProgramData\aos\miniconda3\lib\site-packages\serial\serialwin32.py", line 275, in read
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, '拒绝访问。', None, 5))

yanxiaoyong-ch, 2022-04-25:
从log看和串口没有通信成功，麻烦确认下串口是否正确