alibaba/AliOS-Things #1463: [Bug Report]: 部署运行之后出现这个怎么解决

krisaaaaa, 2022-03-28:
### What happened?

A bug happened!
运行环境:win10  vscode
固件版本:haaaseduk1 python轻应用 温湿度上云

### Version

master (Default)

### What soultions are you seeing the problem on?

eduk1_demo

### Relevant log output

```shell
部署运行后出现下面这个，请问是什么问题嘞
.......
.......
<<< ACK
EOF
>>> EOT
<<< 0x15
>>> EOT
<<< 0x6
<<< 0x43
Packet End >>>
<<< 0x6
File: pyamp.zip
Size: 774Bytes
Ymodem transfer file finish
send cmd exit
PS C:\Users\Admin\workspace\HaaSPython>
```

skylarCai, 2022-03-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

senwangal, 2022-03-29:
> ### Contact Details
> _No response_
>
> ### What happened?
> A bug happened! 运行环境:win10 vscode 固件版本:haaaseduk1 python轻应用 温湿度上云
>
> ### Version
> master (Default)
>
> ### What soultions are you seeing the problem [...]

这看上去log是正常的，已经推送成功了，请问你的问题是什么？
如果需要看log，可以点击haas控件界面的串口勾选haaseduk1的串口，然后选择波特率1500000

closed 2022-04-08 by skylarCai