alibaba/AliOS-Things #1464: [Bug Report]: Haas edu k1测光照强度，出来日志有问题

yc2548672923, 2022-03-29:
### Contact Details

2548672923@qq.com

### What happened?

A bug happened!
运行环境:HAAS studio
固件版本:HaaSPython-HaaSEDUK1-v1.0.2

### Version

master (Default)

### What soultions are you seeing the problem on?

helloworld_demo

### Relevant log output

```shell
版权所有 (C) Microsoft Corporation。保留所有权利。

尝试新的跨平台 PowerShell https://aka.ms/pscore6

PS D:\VS-example\HaaSPython> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python -m serial.tools.miniterm COM3 1500000 --raw
--- Miniterm on COM3  1500000,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
亮度:5
���度:4
���度:4
���度:3
���度:4
���度:3
���度:4
���度:3
���度:3
���度:3
���度:3
���度:4
���度:3
���度:3
���度:3
���度:52
���度:64
```

skylarCai, 2022-03-29:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yc2548672923, 2022-03-29:
人工回复

ethanlcz, 2022-03-29:
> 

您好，请您重启一下设备看看日志是不是还是乱码？
我们刚刚检查代码里边输出的中文前面是没有任何额外的字符的。另外也在VS Code中测试这段代码没有出现乱码的情况。UART有时候会出现乱码的情况，基本都是因为串口工具的问题导致。不过我用的串口工具和你的一样，都是用HaaS-Studio上面的”串口“ 这个工具。
no need report
亮度:275
no need report
亮度:275
no need report
亮度:274
no need report
亮度:273
no need report
亮度:273
no need report
亮度:274
no need report
亮度:273
no need report
亮度:273
no need report
亮度:275
no need report
亮度:275

ethanlcz, 2022-03-29:
> ### Contact Details
> [2548672923@qq.com](mailto:2548672923@qq.com)
> 
> ### What happened?
> A bug happened! 运行环境:HAAS studio 固件版本:HaaSPython-HaaSEDUK1-v1.0.2
> 
> ### Version
> master (Default)
> 
> ### What soultions are you seeing the problem on?
> helloworld_demo
> 
> ### Relevant log output
> ```shell
> 版权所有 (C) Microsoft Corporation。保留所有权利。
> 
> 尝试新的跨平台 PowerShell https://aka.ms/pscore6
> 
> PS D:\VS-example\HaaSPython> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python -m serial.tools.miniterm COM3 1500000 --raw
> --- Miniterm on COM3  1500000,8,N,1 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> 亮度:5
> ���度:4
> ���度:4
> ���度:3
> ���度:4
> ���度:3
> ���度:4
> ���度:3
> ���度:3
> ���度:3
> ���度:3
> ���度:4
> ���度:3
> ���度:3
> ���度:3
> ���度:52
> ���度:64
> ```

您好，请您重启一下设备看看日志是不是还是乱码？
我们刚刚检查代码里边输出的中文前面是没有任何额外的字符的。另外也在VS Code中测试这段代码没有出现乱码的情况。UART有时候会出现乱码的情况，基本都是因为串口工具的问题导致。不过我用的串口工具和你的一样，都是用HaaS-Studio上面的”串口“ 这个工具。
no need report
亮度:275
no need report
亮度:275
no need report
亮度:274
no need report
亮度:273
no need report
亮度:273
no need report
亮度:274
no need report
亮度:273
no need report
亮度:273
no need report
亮度:275
no need report
亮度:275

yc2548672923, 2022-03-29:
对٩(๑^o^๑)۶，还是乱码的，没有改变，您帮我看一下是你图片里的那串代码有问题吗？

  
Contact Details
 
 

 
 
 
 
 
 
 版权所有 (C) Microsoft Corporation。保留所有权利。 尝试新的跨平台 PowerShell https://aka.ms/pscore6 PS D:\VS-example\HaaSPython&gt; $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python -m serial.tools.miniterm COM3 1500000 --raw --- Miniterm on COM3  1500000,8,N,1 --- --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H --- 亮度:5    度:4    度:4    度:3    度:4    度:3    度:4    度:3    度:3    度:3    度:3    度:4    度:3    度:3    度:3    度:52    度:64

ethanlcz, 2022-03-29:
> 您帮我看一下是你图片里的那串代码有问题吗

那串代码我看过了，没有问题的。
你可以加我钉钉好友，我帮你看一下。 钉钉号：ethanlcz

ethanlcz, 2022-04-01:
这个问题可以关闭掉了，串口工具的显示问题。
换用putty工具就没有乱码了。

skylarCai, 2022-04-08:
This issue has not been updated for a while, so we close this issue.  If you have further request, please reopen it or add it.  We will response as soon as possible.

本issue因太久未更新而关闭，如果您还有进一步的问题，请reopen本issue，或者新开一个issue，我们会尽快回复您。

closed 2022-04-08 by skylarCai

yc2548672923, 2022-10-11:
人工支持