alibaba/AliOS-Things #1465: [Feature Request]: M5stack core2 SD卡挂载

rualdeng, 2022-03-29:
### Contact Details

rualdeng@163.com

### What do you want?

M5Stack core2没有开放挂载SD卡接口，我们产品上必须要用，只有自己查资料去写。但还是遇到很多问题。
sdcard = SDCard(slot=2, sck=Pin(18), miso=Pin(38), mosi=Pin(23),  cs=Pin(4), freq=40000000)
uos.mount(sdcard, '/sd')
print(uos.listdir('/sd'))

也查了很多资料和试了很多方式，都是报OSError: [Errno 19] ENODEV。没有识别到卡。麻烦这边帮看下参数设置还有其它什么弄错的。

### Version

master (Default)

skylarCai, 2022-03-29:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yong171966, 2022-03-29:
目前M5Stack Core2上SD卡的功能没有调试，你遇到的问题我们会紧急处理一下。
建议你先用文件系统的方式操作，直接将数据放到/data分区下，等SD卡功能完成后再切换到SD卡上。

closed 2022-04-25 by skylarCai