alibaba/AliOS-Things #1466: [Bug Report]: m5stack core2 OSS上传报bug

rualdeng, 2022-03-30:
### Contact Details

rualdeng@163.com

### What happened?

A bug happened!
运行环境:m5stack core2
固件版本:自行编译m5stack core2的最新版本

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

### Relevant log output

```shell
now time is ===== Wed, 30 Mar 2022 09:34:14 GMT
[makeRequest]response_code: 0
PutObject fail,code:ClientError:200028,message:Timeout was reached.Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds,requestId:

用PC客户端测试了没问题，证明不是OSS的问题。同时OssClientlmpl.cc 中的 ESP_PLATFORM的GMT时间还没改过来。
```

skylarCai, 2022-03-30:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

QingjiangXu, 2022-03-30:
感谢您的反馈。麻烦您确认下：
1. 使用如下链接上的官网固件确认下是否有同样的问题
   https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/startup/M5StackCore2_startup
2. 请问下您是在跑什么应用，测试或者solution出现的错误，并麻烦提供下代码，我们同步来确认？

rualdeng, 2022-03-30:
> 感谢您的反馈。麻烦您确认下：
> 
> 1. 使用如下链接上的官网固件确认下是否有同样的问题
>    https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/startup/M5StackCore2_startup
> 2. 请问下您是在跑什么应用，测试或者solution出现的错误，并麻烦提供下代码，我们同步来确认？

官方的这个固件时间没改过来，直接报时间的错。
def upload():

    ACCESS_KEY = ' '
    ACCESS_SECRET = ' '

        # OSS对象存储配置
    OSS_ENDPOINT = "oss-cn-shenzhen.aliyuncs.com"
    OSS_BUCKET = "xxxxx"
    file = oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, "record.raw")
    print(file)

if  __name__ == '__main__':
    print("what")
    sta_if = network.WLAN(network.STA_IF)
    sta_if.active(True)
    sta_if.connect('MERCURY_511', 'x.29615861')
    # sta_if.connect('denny', 'deng2486')
    utime.sleep_ms(500)
    begin = utime.time()
    while not sta_if.isconnected():
        print("the net is none")
        utime.sleep_ms(500)
        print('this time is')
        print(utime.localtime())

    if sta_if.isconnected():
        print(str(sta_if.isconnected()))
        print("联上网了")
        # power()
        # utime.sleep(1)
        # music()
        # ret = sntp.setTime()
        # while ret != 0:
        #     print("还没有设置好时间")
        # print("设置时间了")
    end = utime.time()
    print('the time is ' ,end - begin)
    # print("the sdcard the ret  ==== " +  str(ret))
    print(uos.listdir('/'))
    utime.sleep(1)
    upload()

rualdeng, 2022-03-30:
> ```shell
> 同时OssClientlmpl.cc 中的 ESP_PLATFORM的GMT时间还没改过来。
> ```

官方固件没用GMT时间，直接报时间的报错。OssClientlmpl.cc 中的 ESP_PLATFORM的GMT时间还没改过来。我已经PR了，TIME - 3600*8，下载最新的源码，这块还没改过来。

rualdeng, 2022-03-30:
> 感谢您的反馈。麻烦您确认下：
> 
> 1. 使用如下链接上的官网固件确认下是否有同样的问题
>    https://haas.iot.aliyun.com/haasapi/index.html#/Python/docs/zh-CN/startup/M5StackCore2_startup
> 2. 请问下您是在跑什么应用，测试或者solution出现的错误，并麻烦提供下代码，我们同步来确认？

官方固件没用GMT时间，直接报时间的报错。OssClientlmpl.cc 中的 ESP_PLATFORM的GMT时间还没改过来。我已经PR了，TIME - 3600*8，下载最新的源码，这块还没改过来。

rualdeng, 2022-03-30:
重新编译后，搞定了。没事了

closed 2022-03-31 by skylarCai