alibaba/AliOS-Things #1470: [Bug Report]: haas edu k1 温湿度上云

krisaaaaa, 2022-04-02:
### What happened?

A bug happened!
运行环境:win10 vscode
固件版本:haas edu k1 python轻应用

### Version

master (Default)

### Relevant log output

```shell
部署运行之后，上传的温湿度数据不对，这是I²C总线驱动或传感器没有配置好吗，怎么解决呢?是不是需要自己编写代码呢

96724/net_wq | cw1200_bss_info_changed=0x10 hw_priv:0x20040368 priv:0x200410b0

    96725/net_wq | BSS_CHANGED_ASSOC.

    96725/net_wq | [STA] DTIM 2, interval: 100

    96725/net_wq | [STA] Preamble: 1, Greenfield: 0, Aid: 1, Rates: 0x00003FC0, Basic: 0x00000540

    96827/net_wq | cw1200_bss_info_changed=0x10 hw_priv:0x20040368 priv:0x200410b0

    96827/net_wq | BSS_CHANGED_ASSOC.

    96828/net_wq | [STA] DTIM 2, interval: 100

    96828/net_wq | [STA] Preamble: 1, Greenfield: 0, Aid: 1, Rates: 0x00003FC0, Basic: 0x00000540

<E><0> [aos_i2c_master_send:479]i2c1 master send failed, ret:0
<E><0> [i2c_device_ioctl:205]i2c1 tx failed, tx_buffer:0x3406a4c0, d->length:1, d->timeout:16777215, ret:0
[  95.017]<E>DRIVER_I2C aos_hal_i2c_master_send failed
<E><0> [aos_i2c_master_send:479]i2c1 master send failed, ret:0
<E><0> [i2c_device_ioctl:205]i2c1 tx failed, tx_buffer:0x3406a570, d->length:1, d->timeout:16777215, ret:0
[  95.049]<E>DRIVER_I2C aos_hal_i2c_master_send failed
uploading data:  {"CurrentTemperature": 128.9, "CurrentHumidity": 119}
```

skylarCai, 2022-04-02:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

krisaaaaa, 2022-04-07:
人工支持

ethanlcz, 2022-04-07:
> 

> 

看日志是I2C总线不同，请问下你运行的是哪个案例呢？案例链接是哪一个？

krisaaaaa, 2022-04-07:
> > 
> 
> > 
> 
> 看日志是I2C总线不同，请问下你运行的是哪个案例呢？案例链接是哪一个？

这是案例链接：https://gitee.com/haasedu/haasedu/blob/release_2.0/1-%E5%BF%AB%E9%80%9F%E5%BC%80%E5%A7%8B/3-%E7%89%A9%E8%81%94%E7%BD%91%E8%AE%BE%E5%A4%87%E4%B8%8A%E4%BA%91/HaaS_EDU_K1_%E6%B8%A9%E6%B9%BF%E5%BA%A6%E4%B8%8A%E4%BA%91.md

krisaaaaa, 2022-04-07:
人工支持

ethanlcz, 2022-04-07:
> > > 
> > 
> > 
> > > 
> > 
> > 
> > 看日志是I2C总线不同，请问下你运行的是哪个案例呢？案例链接是哪一个？
> 
> 这是案例链接：https://gitee.com/haasedu/haasedu/blob/release_2.0/1-%E5%BF%AB%E9%80%9F%E5%BC%80%E5%A7%8B/3-%E7%89%A9%E8%81%94%E7%BD%91%E8%AE%BE%E5%A4%87%E4%B8%8A%E4%BA%91/HaaS_EDU_K1_%E6%B8%A9%E6%B9%BF%E5%BA%A6%E4%B8%8A%E4%BA%91.md

请加我钉钉，我来看看是遇到了什么问题。钉钉号：ethanlcz

skylarCai, 2022-04-08:
请联系钉钉，感谢您的支持

This issue has not been updated for a while, so we close this issue.  If you have further request, please reopen it or add it.  We will response as soon as possible.

本issue因太久未更新而关闭，如果您还有进一步的问题，请reopen本issue，或者新开一个issue，我们会尽快回复您。

closed 2022-04-08 by skylarCai