alibaba/AliOS-Things #1471: [Feature Request]: haas edu k1起夜灯实验传感器总是输出高电平

Ryze-b, 2022-04-03:
### Contact Details

2682394086@qq.com

### What do you want?

没有用官方支持硬件积木的红外释热传感器，选用的是tb上的微型红外释热传感器，感觉两者参数差不多，但是得不出准确的结果，应该如何修改代码呢？

### Version

master (Default)

skylarCai, 2022-04-03:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-04-05:
> ### Contact Details
> [2682394086@qq.com](mailto:2682394086@qq.com)
> 
> ### What do you want?
> 没有用官方支持硬件积木的红外释热传感器，选用的是tb上的微型红外释热传感器，感觉两者参数差不多，但是得不出准确的结果，应该如何修改代码呢？
> 
> ### Version
> master (Default)

请问你用的哪款传感器呢？它的数据手册中在检测到有人和无人的情况默认输出电平各是什么？
你是怎么确认它一直输出的都是高电平呢？

Ryze-b, 2022-04-05:
[image]
淘宝上的risym店，无论那个角度有人无人，获取的值一直是1

Ryze-b, 2022-04-05:
> > ### Contact Details
> > [2682394086@qq.com](mailto:2682394086@qq.com)
> > ### What do you want?
> > 没有用官方支持硬件积木的红外释热传感器，选用的是tb上的微型红外释热传感器，感觉两者参数差不多，但是得不出准确的结果，应该如何修改代码呢？
> > ### Version
> > master (Default)
> 
> 请问你用的哪款传感器呢？它的数据手册中在检测到有人和无人的情况默认输出电平各是什么？ 你是怎么确认它一直输出的都是高电平呢？
[image]

淘宝上的risym店，无论那个角度有人无人，获取的值一直是1

ethanlcz, 2022-04-05:
> > > ### Contact Details
> > > [2682394086@qq.com](mailto:2682394086@qq.com)
> > > ### What do you want?
> > > 没有用官方支持硬件积木的红外释热传感器，选用的是tb上的微型红外释热传感器，感觉两者参数差不多，但是得不出准确的结果，应该如何修改代码呢？
> > > ### Version
> > > master (Default)
> > 
> > 
> > 请问你用的哪款传感器呢？它的数据手册中在检测到有人和无人的情况默认输出电平各是什么？ 你是怎么确认它一直输出的都是高电平呢？
> > [image]
> 
> 淘宝上的risym店，无论那个角度有人无人，获取的值一直是1

人体热释电传感器的反应比较灵敏，只要有人在它的感应范围内稍微动一下，它就可以感应到。
从你现在的现象来看，我建议是先确定你的传感器是否有坏掉，最直接的方法是找万用表来量测传感器输出是一直输出高电平？还是说没有人体移动的时候它会输出低电平。如果它一直输出高电平，则说明是传感器坏掉了。如果传感器在人体完全不动的情况下会输出低电平，说明传感器没有问题。
确认传感器没有问题之后可以将eduk1的检测引脚接到vcc或gnd上看这个引脚读到的输入是否正常。

skylarCai, 2022-04-08:
This issue has not been updated for a while, so we close this issue.  If you have further request, please reopen it or add it.  We will response as soon as possible.

本issue因太久未更新而关闭，如果您还有进一步的问题，请reopen本issue，或者新开一个issue，我们会尽快回复您。

closed 2022-04-08 by skylarCai

Ryze-b, 2022-04-08:
> 人体热释电传感器的反应比较灵敏，只要有人在它的感应范围内稍微动一下，它就可以感应到。 从你现在的现象来看，我建议是先确定你的传感器是否有坏掉，最直接的方法是找万用表来量测传感器输出是一直输出高电平？还是说没有人体移动的时候它会输出低电平。如果它一直输出高电平，则说明是传感器坏掉了。如果传感器在人体完全不动的情况下会输出低电平，说明传感器没有问题。 [...]

补发了两个，结果测得的无感应电压是2.7V，高电平是3.3V。。

> 人体热释电传感器的反应比较灵敏，只要有人在它的感应范围内稍微动一下，它就可以感应到。 从你现在的现象来看，我建议是先确定你的传感器是否有坏掉，最直接的方法是找万用表来量测传感器输出是一直输出高电平？还是说没有人体移动的时候它会输出低电平。如果它一直输出高电平，则说明是传感器坏掉了。如果传感器在人体完全不动的情况下会输出低电平，说明传感器没有问题。 [...]

补发了两个，结果测得的无感应电压是2.7V，高电平是3.3V。。

Ryze-b, 2022-04-08:
补发了两个，结果测得的无感应电压是2.7V，高电平是3.3V。。

Ryze-b, 2022-04-08:
> 人体热释电传感器的反应比较灵敏，只要有人在它的感应范围内稍微动一下，它就可以感应到。 从你现在的现象来看，我建议是先确定你的传感器是否有坏掉，最直接的方法是找万用表来量测传感器输出是一直输出高电平？还是说没有人体移动的时候它会输出低电平。如果它一直输出高电平，则说明是传感器坏掉了。如果传感器在人体完全不动的情况下会输出低电平，说明传感器没有问题。 [...]

补发了两个，结果测得的无感应电压值是2.7V，代码里估计是0V才触发关闭。。
这是商家的产品问题吗？还是说代码不兼容..