alibaba/AliOS-Things #1472: [Bug Report]: haaseduk1

krisaaaaa, 2022-04-04:
### What happened?

A bug happened!
运行环境:
固件版本:

### Version

master (Default)

### Relevant log output

```shell
Haaseduk1开发板之前做了一个温湿度上云的项目，现在开机后只有电源指示灯会亮，OLED屏幕不亮也不显示什么东西，一开始的是十个案例也不显示了。请问这要怎么解决呢，可以重置开发板吗，还是要怎么操作呢
```

skylarCai, 2022-04-04:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

krisaaaaa, 2022-04-05:
是的，烧录了自己的固件。请问要怎么恢复成初始状态呢

ethanlcz, 2022-04-25:
> 是的，烧录了自己的固件。请问要怎么恢复成初始状态呢
可以回复称初始状态的，稍等我们找一篇说明文章。

ethanlcz, 2022-04-25:
> 是的，烧录了自己的固件。请问要怎么恢复成初始状态呢

请参考 HaaS EDU K1硬件积木中“恢复出厂设置”章节的说明下载出厂固件并进行固件烧录。
https://haas.iot.aliyun.com/solution/detail/hardware?versionId=800C5AB3B8A4A88800000001&dataId=800C5AB3B8A4A888

closed 2022-04-27 by skylarCai