alibaba/AliOS-Things #1473: [Feature Request]: haas100板的WiFi连接不上

HRO547, 2022-04-06:
### Contact Details

2594100025@qq.com

### What do you want?

haas100在连接WiFi上云过程中出现一下串口日志（连接不稳定，多WiFi测试只能连接到一个WiFi上云
[New File.txt](https://github.com/alibaba/AliOS-Things/files/8427015/New.File.txt)
）

### Version

master (Default)

skylarCai, 2022-04-06:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

HRO547, 2022-04-06:
然后后面就重复这段日志
 89187/net_wq | cw1200_bss_info_changed=0x12 hw_priv:0x20038fd8 priv:0x20039d20

    89187/net_wq | BSS_CHANGED_ASSOC.

    89187/net_wq | [STA] Non-GF STA present

    89187/net_wq | [STA] DTIM 2, interval: 100

    89187/net_wq | [STA] Preamble: 1, Greenfield: 0, Aid: 12, Rates: 0x00003FC0, Basic: 0x00000540

YiluMao, 2022-04-06:
我们看下

yanxiaoyong-ch, 2022-04-06:
1.请检查天线是否接上；2.如果天线接上，请将当前路由器型号发下，并更换个路由器试试

closed 2022-04-19 by skylarCai