alibaba/AliOS-Things #1475: [Bug Report]:

TangYuFan, 2022-04-07:
### Contact Details

995435702@qq.com

### What happened?

A bug happened!
运行环境:
haas edu k1 
固件版本:
HaaSPython-HaaSEDUK1-v1.0.2.zip
示例代码：

#定义mqtt订阅的回调函数
def mqtt_callback(topic,payload):
    t = topic.decode("utf-8")
    p = payload.decode("utf-8")
    print("on_subcribe: topic is:" + t)
    print("on_subcribe: payload is:" + p)
    #如果是rpc则进行响应

# 链接mqtt服务端
def connect_tb():
    global client 
    client = mqtt.MQTTClient(client_id=deviceName,user=deviceName,password=deviceToken,server=mqttServerHost,port=mqttServerPort)
    client.connect()
    #订阅回调函数
    client.set_callback(mqtt_callback)
    #订阅属性更新
    client.subscribe(topic=topic1,qos=1)
    #订阅rpc
    client.subscribe(topic=topic2,qos=1)

# 发送遥测
def telemetryReport(payload):
    client.publish(topic=topic3,msg=payload)
    
# 发送属性
def attributesReport(payload):
    client.publish(topic=topic4,msg=payload)

# 定时监听mqtt订阅数据
def waitMsg(delay,id):
    while True:
        client.wait_msg()

if __name__ == '__main__':

    #初始化led
    oled_init()

    #链接wifi
    get_wifi_status()

    #初始化光照
    ap3216c_init()

    #链接tb
    connect_tb()

    #---------执行数据监听-------
    _thread.start_new_thread(waitMsg,("x","b"))

    while True:
        #遥测上报
        payload = {'light':lightness}
        telemetryReport(ujson.dumps(payload))
        utime.sleep(1)

### Version
[... 313 lines cut ...]
(0x3405B558): 0x340610F2 0x2C27869C 0x3405B62C 0x00000000
(0x3405B568): 0x00000000 0x0C139861 0x00000002 0x3405B618
(0x3405B578): 0x3405B640 0x3405B63C 0x00000002 0x00000000
(0x3405B588): 0x3405B634 0x00000016 0x2C275B34 0x3405B68C
(0x3405B598): 0x3405B618 0x00000000 0x3405B618 0x3473452C
(0x3405B6D8): 0x3405B700 0x00000001 0x00000002 0x340567C0
(0x3405B6E8): 0x0000057D 0x3405B700 0x3405B700 0x0C08E367
(0x3405B6F8): 0x00000005 0x3405E980 0x7461642F 0x79702F61
(0x3405B708): 0x2F706D61 0x6E69616D 0x0079702E 0x00000000
.........................( All Zeros ).........................
(0x3405B7F8): 0x00000000 0x00000000 0x34056798 0x05050505
(0x3405B808): 0x06060606 0x07070707 0x08080808 0x0C08E413
(0x3405B818): 0x34056798 0x01010101 0x02020202 0x03030303
(0x3405B828): 0x12121212 0x04040404 0x05050505 0x00235149
!!!!!!!!!! dump end   !!Exception in thread rx:
Traceback (most recent call last):
  File "C:\ProgramData\aos\miniconda3\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\ProgramData\aos\miniconda3\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\ProgramData\aos\miniconda3\lib\site-packages\serial\tools\miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "C:\ProgramData\aos\miniconda3\lib\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
```

skylarCai, 2022-04-07:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

TangYuFan, 2022-04-09:
已经解决

closed 2022-04-09 by TangYuFan