alibaba/AliOS-Things #1476: [Bug Report]:

TangYuFan, 2022-04-08:
### Contact Details

995435702@qq.com

### What happened?

A bug happened!
运行环境:
haas edu k1
固件版本:
HaaSPython-HaaSEDUK1-v1.0.2.zip

### Version

master (Default)

### What soultions are you seeing the problem on?

eduk1_demo

### Relevant log output

```shell
from driver import I2C

def si7006_init():
    si7006Obj = I2C()
    si7006Obj.open("si7006")
    global humitureDev
    humitureDev = SI7006(si7006Obj)
    version = humitureDev.getVer()
    chipID = humitureDev.getID()
    print("si7006_init ver:",version,",chipID:",chipID)

def get_temp_humi():
    global temperature,humidity
    humniture = humitureDev.getTempHumidity()          
    temperature = humniture[0]                          
    humidity = humniture[1]                            
    print("temperature:",temperature,"humidity:",humidity)    

if __name__ == '__main__':

    si7006_init()

    while True:
        #温湿度
        get_temp_humi()
        utime.sleep(1)

错误日志：
[  10.331]<E>DRIVER_I2C aos_hal_i2c_master_recv failed
si7006_init ver: 0 ,chipID: bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
[  10.562]<E>DRIVER_I2C aos_hal_i2c_master_send failed
[  10.592]<E>DRIVER_I2C aos_hal_i2c_master_recv failed
Traceback (most recent call last):
  File "/data/pyamp/main.py", line 268, in <module>
  File "/data/pyamp/main.py", line 129, in get_temp_humi
  File "/data/pyamp/si7006.py", line 132, in getTempHumidity
  File "/data/pyamp/si7006.py", line 100, in getTemperature
```

skylarCai, 2022-04-08:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-04-08:
> ### Contact Details
> [995435702@qq.com](mailto:995435702@qq.com)
> 
> ### What happened?
> A bug happened! 运行环境: haas edu k1 固件版本: HaaSPython-HaaSEDUK1-v1.0.2.zip
> 
> ### Version
> master (Default)
> 
> ### What soultions are you seeing the problem on?
> eduk1_demo
> 
> ### Relevant log output
> ```shell
> from driver import I2C
> 
> def si7006_init():
>     si7006Obj = I2C()
>     si7006Obj.open("si7006")
>     global humitureDev
>     humitureDev = SI7006(si7006Obj)
>     version = humitureDev.getVer()
>     chipID = humitureDev.getID()
>     print("si7006_init ver:",version,",chipID:",chipID)
> 
> def get_temp_humi():
>     global temperature,humidity
>     humniture = humitureDev.getTempHumidity()          
>     temperature = humniture[0]                          
>     humidity = humniture[1]                            
>     print("temperature:",temperature,"humidity:",humidity)    
> 
> 
> if __name__ == '__main__':
> 
>     si7006_init()
> 
>     while True:
>         #温湿度
>         get_temp_humi()
>         utime.sleep(1)
> 
> 
> 错误日志：
> [  10.331]<E>DRIVER_I2C aos_hal_i2c_master_recv failed
> si7006_init ver: 0 ,chipID: bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
> [  10.562]<E>DRIVER_I2C aos_hal_i2c_master_send failed
> [  10.592]<E>DRIVER_I2C aos_hal_i2c_master_recv failed
> Traceback (most recent call last):
>   File "/data/pyamp/main.py", line 268, in <module>
>   File "/data/pyamp/main.py", line 129, in get_temp_humi
>   File "/data/pyamp/si7006.py", line 132, in getTempHumidity
>   File "/data/pyamp/si7006.py", line 100, in getTemperature
> ```

咱们的HaaS EDU K1有两个版本，两个版本的温湿度传感器SI7006和CHT8305。出现这个错误大概率可能是因为你的K1里边是CHT8305传感器。
可以运行下这个案例里边的代码，来看看是否可以运行成功。
https://haas.iot.aliyun.com/solution/detail/solution?versionId=800AC5B00B951C5300000001&dataId=800AC5B00B951C53

TangYuFan, 2022-04-08:
我用的就是这个例子里面的代码，切换成 cth8305.py 也是同样的问题：
# from si7006 import SI7006
from cht8305 import CHT8305

# cht8305温湿度传感器初始化
def cht8305Init():
    print("cht8305Init")
    i2cObj = I2C()
    i2cObj.open("cht8305")
    humitureDev = CHT8305(i2cObj)
    temperature = humitureDev.getTemperature()
    humidity = humitureDev.getHumidity()
    print("temperature:",temperature,"humidity:",humidity)

错误日志：
     5450/py_engine | spi 1 open succ
oledInit success
ap3216c success
cht8305Init
[   4.513]<E>DRIVER_I2C aos_hal_i2c_master_send failed
[   4.544]<E>DRIVER_I2C aos_hal_i2c_master_recv failed
Traceback (most recent call last):
  File "/data/pyamp/main.py", line 182, in <module>
  File "/data/pyamp/main.py", line 79, in cht8305Init
  File "/data/pyamp/cht8305.py", line 52, in getTemperature

TangYuFan, 2022-04-09:
人工支持

ethanlcz, 2022-04-09:
> 我用的就是这个例子里面的代码，切换成 cth8305.py 也是同样的问题：
> 
> # from si7006 import SI7006
> from cht8305 import CHT8305
> 
> # cht8305温湿度传感器初始化
> def cht8305Init(): print("cht8305Init") i2cObj = I2C() i2cObj.open("cht8305") humitureDev = CHT8305(i2cObj) temperature = humitureDev.getTemperature() humidity = humitureDev.getHumidity() print("temperature:",temperature,"humidity:",humidity)
> 
> 错误日志： 5450/py_engine | spi 1 open succ oledInit success ap3216c success cht8305Init [ 4.513]DRIVER_I2C aos_hal_i2c_master_send failed [ 4.544]DRIVER_I2C aos_hal_i2c_master_recv failed Traceback (most recent call last): File "/data/pyamp/main.py", line 182, in File "/data/pyamp/main.py", line 79, in cht8305Init File "/data/pyamp/cht8305.py", line 52, in getTemperature

麻烦加我钉钉，我来看下是什么问题。钉钉号：ethanlcz

closed 2022-04-19 by skylarCai