alibaba/AliOS-Things #1477: edu k1 起夜灯实验案例遇到的小问题

Ryze-b, 2022-04-08:
> > > > ### Contact Details
> > > > [2682394086@qq.com](mailto:2682394086@qq.com)
> > > > ### What do you want?
> > > > 没有用官方支持硬件积木的红外释热传感器，选用的是tb上的微型红外释热传感器，感觉两者参数差不多，但是得不出准确的结果，应该如何修改代码呢？
> > > > ### Version
> > > > master (Default)
> > > 
> > > 
> > > 请问你用的哪款传感器呢？它的数据手册中在检测到有人和无人的情况默认输出电平各是什么？ 你是怎么确认它一直输出的都是高电平呢？
> > 
> > 
> > 淘宝上的risym店，无论那个角度有人无人，获取的值一直是1
> 
> 人体热释电传感器的反应比较灵敏，只要有人在它的感应范围内稍微动一下，它就可以感应到。 从你现在的现象来看，我建议是先确定你的传感器是否有坏掉，最直接的方法是找万用表来量测传感器输出是一直输出高电平？还是说没有人体移动的时候它会输出低电平。如果它一直输出高电平，则说明是传感器坏掉了。如果传感器在人体完全不动的情况下会输出低电平，说明传感器没有问题。 确认传感器没有问题之后可以将eduk1的检测引脚接到vcc或gnd上看这个引脚读到的输入是否正常。

补发了两个，结果测得的无感应低电平电压是2.7V。。
这是商家的产品问题吗？还是说代码不兼容..

_Originally posted by @Ryze-b in https://github.com/alibaba/AliOS-Things/issues/1471#issuecomment-1092823944_

skylarCai, 2022-04-08:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

senwangal, 2022-04-08:
1、获取值为负1，说明跟传感器已经没有关系，纯粹是接口使用错误。
另外请贴一些你的一些log，关键代码。确认下是否有问题。
参考案例确保接线没有接错，确认下board.json文件有没有配置正确。

2、另外淘宝链接可以贴一下吗，我看一下参数，看你的描述像是用的ADC获取电压值，这类人体感应跟案例中用的返回0/1值的读法是不一样的。需要参考比如燃气监测相关的案例。

Ryze-b, 2022-04-08:
> 1、获取值为负1，说明跟传感器已经没有关系，纯粹是接口使用错误。 另外请贴一些你的一些log，关键代码。确认下是否有问题。 参考案例确保接线没有接错，确认下board.json文件有没有配置正确。
> 
> 2、另外淘宝链接可以贴一下吗，我看一下参数，看你的描述像是用的ADC获取电压值，这类人体感应跟案例中用的返回0/1值的读法是不一样的。需要参考比如燃气监测相关的案例。

使用的是edu k1的官方案例，输出为1即一直是高电平，后面万用表测了一下是有人时为3.3V，无人时为2.7V。
[image]

【淘宝】https://m.tb.cn/h.fLzZQOh?tk=Uw9f2SlBh5e「微型人体红外感应模块 PIR红外热释电感应开关探头/菲涅尔透镜」
点击链接直接打开

senwangal, 2022-04-12:
案例中用到的是中断传感器，GPIO 0或者1，相应的电压值就是0跟3.3左右，如你实际测量到，3.3v或者2.7V，反馈到GPIO肯定都是1.
对应你的传感器是不能直接讨论这个案例的，需要用到ADC接口来读电压范围，或者换用我们案例中的传感器

closed 2022-04-19 by skylarCai