alibaba/AliOS-Things #1478: [ 164.216]<E>DRIVER_I2C aos_hal_i2c_master_send failed [ 164.246]<E>DRIVER_I2C aos_hal_i2c_master_recv failed[Bug Report]:

TangYuFan, 2022-04-08:
### Contact Details

995435702@qq.com

### What happened?

急急急要交作业!!!!
运行环境:
HaaSEDUK1 / Windows / VsCode
固件版本:
HaaSPython-HaaSEDUK1-v1.0.2.zip
https://hli.aliyuncs.com/p/config/HaaS_Python/HaaSPython-HaaSEDUK1-v1.0.2.zip

### Version

master (Default)

### What soultions are you seeing the problem on?

eduk1_demo

### Relevant log output

```shell
我运行了多个i2c示例，包括光照(ap3216c)、温湿度(si7006/cht8305)、陀螺仪(mpu6050)、气压(spl06)。仅仅光照ap3216c能够正常的读书，其他示例都异常：
[ 164.216]<E>DRIVER_I2C aos_hal_i2c_master_send failed
[ 164.246]<E>DRIVER_I2C aos_hal_i2c_master_recv failed

其中ap3216c.py、cht8305.py、haaseduk1.py、mpu6050.py、sh1106.py、si7006.py、spl06.py几个文件均来自于 alios-things源码：
https://gitee.com/alios-things-admin/AliOS-Things/tree/master/components/py_engine/framework

代码如下：
from haaseduk1 import HAASEDUK1   
import ap3216c  
from mpu6050 import MPU6050 
from spl06 import SPL06 

def ap3216cInit():
    global ap3216cDev
    i2cDev = I2C()
    i2cDev.open("ap3216c")
    ap3216cDev = ap3216c.AP3216C(i2cDev)
    print("ap3216c init finished")

def si7006Andcht8305Init():
    board = HAASEDUK1()               
    hwID = board.getHWID()              
    i2cObj = I2C()
    global humitureDev
    if (hwID == 1):
        from cht8305 import CHT8305      
        i2cObj.open("cht8305")           
        humitureDev = CHT8305(i2cObj)  
        print("cht8305 inited!")
    else:
        from si7006 import SI7006       
        i2cObj.open("si7006")            
        humitureDev = SI7006(i2cObj)   
        version = humitureDev.getVer() 
        chipID = humitureDev.getID()   
        print("si7006 inited ! version:%d, chipID:%d" , version, chipID)

def mpu6050Init():
    global mpu6050Dev
    mpu6050Dev = MPU6050()
    mpu6050Dev.open("mpu6050")
    mpu6050Dev.init()
    print("mpu6050Init success")

def spl06Init():
    global spl06Dev
    spl06Dev = SPL06()
    spl06Dev.open("spl06")
    spl06Dev.init()
    print("spl06Init success")

def sdechulReport(threadId):
    while True:
        #获取光照
        light = ap3216cDev.getIlluminance()
        #获取 si7006 温度/湿度
        temperature = humitureDev.getTemperature()          
        humidity = humitureDev.getHumidity()
        #获取 mpu6050 陀螺仪原始数据
        ac = mpu6050Dev.get_Accelerometer()
        gy = mpu6050Dev.get_Gyroscope()
        tm = mpu6050Dev.get_Temperature()
        print("ac[0]",ac[0],"ac[1]",ac[1],"ac[2]",ac[2])
        print("gy[0]",gy[0],"gy[1]",gy[1],"gy[2]",gy[2])
        #获取spl06气压/温度数据
        spl = str(spl06Dev.spl06_getdata())
        print("spl:",spl)
        utime.sleep(3)

if __name__ == '__main__':

    utime.sleep(5)

    ap3216cInit()
    si7006Andcht8305Init()
    mpu6050Init()
    spl06Init()

    #设置栈大小 默认8848字节
    _thread.stack_size(17696)

    # 创建线程 定时访问传感器获取数据
    _thread.start_new_thread(sdechulReport, ["thread2"])

    while True:
        utime.sleep(10000000)
```

TangYuFan, 2022-04-08:
board.js文件如下：
{
  "name": "haasedu",
  "version": "1.0.0",
  "io": {
    "oled_spi": {
      "type": "SPI",
      "port": 1,
      "mode": "master",
      "freq": 26000000
    },
    "oled_dc": {
      "type": "GPIO",
      "port": 28,
      "dir": "output",
      "pull": "pullup"
    },
    "oled_res": {
      "type": "GPIO",
      "port": 30,
      "dir": "output",
      "pull": "pullup"
    },
    "ap3216c": {
      "type": "I2C",
      "port": 1,
      "addrWidth": 7,
      "freq": 100000,
      "mode": "master",
      "devAddr": 30
    },
    "cht8305": {
      "type": "I2C",
      "port": 1,
      "addrWidth": 7,
      "freq": 1000000,
      "mode": "master",
      "devAddr": 64
    },
    "si7006": {
      "type": "I2C",
      "port": 1,
      "addrWidth": 7,
      "freq": 400000,
      "mode": "master",
      "devAddr": 64
    },
    "led1": {
      "type": "GPIO",
      "port": 36,
      "dir": "output",
      "pull": "pullup"
    },
    "led2": {
      "type": "GPIO",
      "port": 35,
      "dir": "output",
      "pull": "pullup"
    },
    "led3": {
      "type": "GPIO",
      "port": 34,
      "dir": "output",
      "pull": "pullup"
    },
    "mpu6050": {
      "type": "I2C",
      "port": 1,
      "addrWidth": 7,
      "freq": 100000,
      "mode": "master",
      "devAddr": 105
    },
    "spl06": {
      "type": "I2C",
      "port": 1,
      "addrWidth": 7,
      "freq": 400000,
      "mode": "master",
      "devAddr": 119
    },
    "qmc5883": {
      "type": "I2C",
      "port": 1,
      "addrWidth": 7,
      "freq": 400000,
      "mode": "master",
      "devAddr": 13
    }
  },
  "debugLevel": "ERROR",
  "repl": "disable"
}

skylarCai, 2022-04-08:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

TangYuFan, 2022-04-09:
人工支持

YiluMao, 2022-04-09:
我们看下

ethanlcz, 2022-04-09:
> 人工支持

咱们线下处理

closed 2022-04-19 by skylarCai