alibaba/AliOS-Things #1479: [Feature Request]: HaaS蓝牙配网问题

zouw96, 2022-04-09:
### Contact Details

zouw96@163.com

### What do you want?

feature request
目前蓝牙配网只有获取wifi密码的功能，alios things的库是提供了三元组的获取，这一部分是不是没有移植过来，实际应用很需要！

### Version

rel_3.3.0 (Main)

skylarCai, 2022-04-09:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-04-11:
请问您想获取三元组的需求是什么？

zouw96, 2022-04-12:
使用python开发。C好像没有移植好的，是不是只能使用alios things自己去移植？

ElliottSeer, 2022-04-25:
> 使用python开发。C好像没有移植好的，是不是只能使用alios things自己去移植？

python开发可以参考  https://github.com/alibaba/AliOS-Things/blob/master/components/py_engine/adapter/esp32/fs/lib/oneMinuteOnCloud.py
可以仿照这里的协议来实现三元组的传输 

但是 不建议量产产品中使用此代码，传输的安全性得不到保证
量产产品中建议使用 [一型一密](https://help.aliyun.com/document_detail/74006.htm?spm=a2c4g.11186623.0.0.668a6379qynWVD#task-m1l-zqq-wfb) 动态注册方式分配三元组

closed 2022-04-27 by skylarCai