alibaba/AliOS-Things #1486: [Bug Report]: HaaS600 很大概率上云失败 即使上云成功了  reset后也有很大概率连不上

ElliottSeer, 2022-04-12:
### Contact Details

钉钉 1jf-jhbi0muows

### What happened?

A bug happened!
运行环境:
HaaS600
固件版本:

### Version

rel_3.3.0 (Main)

### Relevant log output

```shell
get operator name task
Loading the CA root certificate
ok (e skipped)
start prepare client cert
start mbedtls_pk_parse_k????~~
Connecting to/iot-auth.cn-shanghai.aliyuncs.com/443.
D: 55 amp_tcp_establish() | establish tep connection with
server(host=iot-auth.cn-shanghai.aliyuncs.com port=443)
D: 189 amp_tcp_establish() | success to establish tcp, fd=3
ok
• Setting up the SSL/TLS structure.
ok
Performing the SSL/TLS handshake.
ok
Verifying peer X.509 certificate.
certificate verification result: oxe
need release client crt&key
ssl disconnect
```

skylarCai, 2022-04-12:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

qisheng-seu, 2022-04-27:
您好，请提供下以下信息方便我们技术同学排查，
1、HaaS600的硬件版本，可以从主芯片的丝印信息获取：EC100Y ? EC600S ? 
2、连接云失败的完整串口日志。

谢谢

qisheng-seu, 2022-04-27:
亲，我们内部技术同学反馈您用的是100Y，已经通过升级固件帮您解决该问题了。感谢您的参与

closed 2022-04-27 by skylarCai