alibaba/AliOS-Things #1487: krhino_intrpt_enter使用ulog log_get_mutex返回RHINO_NOT_CALLED_BY_INTRPT

RiderChen, 2022-04-13:
如题
请问下，中断中使用ulog在alios中是被禁止的么？
或者说有其他log方式推荐？个人偏向于linux printk；
谢谢

skylarCai, 2022-04-13:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

jinliangli, 2022-04-14:
中断中使用ulog在alios中是被禁止的么？
 -- 是的， 因为中断上下文中不能使用任何可能引起阻塞的操作。所以ulog/printf都是不能使用。

请使用alios things的printk，使用时请确保debug组件被依赖，在使用时包含aos/debug.h即可。

用法与printf相似， 如printk("%s:%d\r\n", __func__, __LINE__);

RiderChen, 2022-04-14:
用ulog的需求主要是想要保存这一段log到sd卡中；
开启ULOG_CONFIG_ASYNC、LOG_DIR_ASYNC，按理说只是使用aos_queue_send，而且进入中断前加上了krhino_intrpt_enter，disable sched ；这种在ulog也不允许么？
那要实现这种需求，有其他办法？

jinliangli, 2022-04-14:
ulog 中的log_get_mutex， mutex本身是不能用于中断的。

这种需求，可以放到线程里做， 放到线程有多种方式， 可以看一下内核中的workqueue, 当然也可以自己实现这种线程化。

closed 2022-04-18 by jinliangli