alibaba/AliOS-Things #1488: [Feature Request]: Haas edu板结合了ap3216c和si7006两个i2c模块，结果ap3216c的i2c信息读取失败

Ryze-b, 2022-04-13:
### Contact Details

2682394086@qq.com

### What do you want?

[image]
如图，温湿度是正常的，光照强度那边read failed，是不能同时用两个i2c协议的模块吗？

### Version

master (Default)

skylarCai, 2022-04-13:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

senwangal, 2022-04-14:
不同的II2C地址，是可以共存，这个问题请你先看下单独的光照是否可以，确认下是否连线有问题

wstong999, 2022-04-14:
问题已经收到，我们这边尽快排查解决，有消息会第一时间通知您

Ryze-b, 2022-04-14:
> 不同的II2C地址，是可以共存，这个问题请你先看下单独的光照是否可以，确认下是否连线有问题

单独的光照是可以的。

Ryze-b, 2022-04-15:
[image]
刚刚测了一下同个程序，把测温湿度的代码注释掉之后光照可以正常检测到了。
还有不知为何会反复出现断开重连这一现象

Ryze-b, 2022-04-16:
[image]
补充点日志...

Ryze-b, 2022-04-17:
哦，测了一下发现我这块板测温湿度调用的是cht8305模块，不是si7006
[image]
Edu K1C和K1是两个不同版本的吗？

senwangal, 2022-04-19:
K1C 跟K1的传感器是不一样的，cht8305用到的是K1C

Ryze-b, 2022-04-19:
明天我用sht3x试试，如果可以用的话再更新后续。

Ryze-b, 2022-04-24:
sht3x单独是可以用的，顶替板载的cht8305，但也是无法和板载的AP3216C传感器一起使用。还是一样的问题，第二个调用的I2C协议的传感器无法写入和读取数据。

ethanlcz, 2022-04-27:
> sht3x单独是可以用的，顶替板载的cht8305，但也是无法和板载的AP3216C传感器一起使用。还是一样的问题，第二个调用的I2C协议的传感器无法写入和读取数据。

这个问题我们内部正在解决中的。

closed 2022-04-27 by skylarCai