alibaba/AliOS-Things #1489: [Bug Report]: m5报内存泄露

rualdeng, 2022-04-14:
### Contact Details

rualdeng@163.com

### What happened?

A bug happened!
运行环境:
固件版本:

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

### Relevant log output

```shell
设备上云订阅topic，收到订阅的信息后在on_subscribe里进行OSS上传（开个子线程）。然后就报内存泄漏。***ERROR*** A stack overflow in task mp_thread has been detected.

Backtrace:0x40093dd1:0x3fff6680 0x400943bd:0x3fff66a0 0x4009467d:0x3fff66c0 0x40095760:0x3fff6740 0x40094770:0x3fff6780 0x40094726:0x00000000 |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...

下面是重启后再执行一次报的错。

CORRUPT HEAP: multi_heap.c:439 detected at 0x3fff6a78

abort() was called at PC 0x40097e3f on core 0

Backtrace:0x40093dd1:0x3fff5f20 0x400943bd:0x3fff5f40 0x400989d1:0x3fff5f60 0x40097e3f:0x3fff5fd0 0x40098209:0x3fff5ff0 0x400986c4:0x3fff6020 0x40085e7e:0x3fff6040 0x40085ea2:0x3fff6070 0x400989d9:0x3fff6090 0x4032c0d5:0x3fff60b0 0x4033877f:0x3fff60d0 0x40285a47:0x3fff60f0 0x40285a87:0x3fff6120 0x40286fec:0x3fff6140 0x4029373d:0x3fff7440 0x4027fe6e:0x3fff7490 0x4027cca4:0x3fff74d0 0x40281854:0x3fff75d0 0x4028195a:0x3fff7670 0x4027d47a:0x3fff7720 0x4027d7e7:0x3fff7810 0x4027852d:0x3fff7940 0x402805e2:0x3fff7960 0x401140e0:0x3fff7cd0 0x400ef0d9:0x3fff7d00 0x400f79ad:0x3fff7d30 0x400f7afe:0x3fff7d50 0x400fa969:0x3fff7d70 0x400ef164:0x3fff7e10 0x400f79ad:0x3fff7e40 0x400fa8ba:0x3fff7e60 0x400ef164:0x3fff7f00 0x400f79ad:0x3fff7f60 0x400ea477:0x3fff7f80 0x4010b5c1:0x3fff8020

ELF file SHA256: 0000000000000000

Rebooting...

def upload():
    #global filename
    ACCESS_KEY = ''
    ACCESS_SECRET = ''
        # OSS对象存储配置
    OSS_ENDPOINT = ""
   
    OSS_BUCKET = ""
        # test.jpg存放在“/”目录下
    begin = time.time()
    file = oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, '1.amr')
    end = time.time()
    print('the time in upload() ' ,end - begin)

    return file

def on_subscribe(data):
    top = ujson.loads(data['payload'])
    print(top['TargetDevice'])
    if (top["record"] == '1'):
        try:
            _thread.start_new_thread(upload,())
        except Exception as e:
            print(e)

这是什么问题？
```

skylarCai, 2022-04-14:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yong171966, 2022-04-15:
@rualdeng，你编译的固件是alios-things开源的代码？还是你基于alios-thigns修改的后代码？

rualdeng, 2022-04-15:
> @rualdeng，你编译的固件是alios-things开源的代码？还是你基于alios-thigns修改的后代码？

原代码，没有修改过

rualdeng, 2022-04-15:
> @rualdeng，你编译的固件是alios-things开源的代码？还是你基于alios-thigns修改的后代码？

修改那两次编译后，还是一样。Memory allocation failed，但上句用gc.mem_free()查内存还有992528

yong171966, 2022-04-15:
CONFIG_MBEDTLS_IRAM_8BIT_MEM_ALLOC=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=4
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=8
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=8
CONFIG_LWIP_TCP_RECVMBOX_SIZE=8
CONFIG_LWIP_UDP_RECVMBOX_SIZE=8

增加上面的配置信息可以节省wifi占用的internal-8bit ram。可以解决你的问题。

不要使能UI功能，UI占用内存较多。

rualdeng, 2022-04-16:
> 
这个设置是否可以再设置大些，WIFI和LP启动时占用70多K的内存，还是蛮大的。还有就是OSS疑似泄露的问题，还帮忙核查处理一下。谢谢！
没有使用UI功能，开启UI功能，连接WIFI就直接内存不够，崩溃重启了。

rualdeng, 2022-04-19:
这个问题已经解决了，目前剩下的问题是oss内存泄漏问题

closed 2022-04-25 by skylarCai