alibaba/AliOS-Things #1490: [Feature Request]: edu上使用ds18b20传感器出现问题

Ryze-b, 2022-04-16:
### Contact Details

2682394086@qq.com

### What do you want?

[image]
[image]

### Version

master (Default)

skylarCai, 2022-04-16:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-04-16:
> ### Contact Details
> [2682394086@qq.com](mailto:2682394086@qq.com)
> 
> ### What do you want?
> [image] [image]
> 
> ### Version
> master (Default)

亲，这个问题我们验证一下回复您，谢谢。

ethanlcz, 2022-04-18:
> ### Contact Details
> [2682394086@qq.com](mailto:2682394086@qq.com)
> 
> ### What do you want?
> [image] [image]
> 
> ### Version
> master (Default)

确实是存在这个问题，我们在内部提问题单进行跟踪，谢谢您的反馈。

closed 2022-04-25 by skylarCai