alibaba/AliOS-Things #1491: [Bug Report]: HaaS Studio v2.0.0 使用串口报错

ElliottSeer, 2022-04-17:
### Contact Details

钉钉 92m_9lis6vzqt

### What happened?

A bug happened!
运行环境: windows
固件版本: HaaS Studio v2.0.0

[image]

### Version

rel_3.3.0 (Main)

skylarCai, 2022-04-17:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

senwangal, 2022-04-19:
从log来看是conda active失败，请提供下你的系统版本信息，

senwangal, 2022-04-19:
这个操作是在什么情况下触发，部署运行还是点击串口？能否提供下相关的log

closed 2022-05-06 by skylarCai