alibaba/AliOS-Things #1492: [Bug Report]: OSS内存泄露

rualdeng, 2022-04-19:
### Contact Details

rualdeng@163.com

### What happened?

A bug happened!
运行环境:
固件版本:最新版本
[image]

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

### Relevant log output

```shell
OSS上传每次都会消耗700字节的内存。存在内存泄露，帮忙分析查询下是什么原因，谢谢！因为之前的OSS上传，没有路径可选，只能上传根目录文件，很不方便。所以我自行改成了oss_upload_file这个函数，和增加了一个可选路径的参数，其它没变。所以请分析下oss_upload_file这个函数的泄露。谢谢！
![27EC9A2F-3FBF-4756-8B3E-351781F14004](https://user-images.githubusercontent.com/19386892/163943825-e4f6a30b-1d76-4f90-b3c7-4d4dc824b357.png)
```

skylarCai, 2022-04-19:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

rualdeng, 2022-04-19:
同时每隔几秒（5-10秒）上传，就会很慢。20K的要20多秒，正常1-2秒就可上传。因业务需要可能要频繁上传（每隔5-10秒上传20K左右）。所以上传的稳定性也是非常重要的。目前是这两个问题

YiluMao, 2022-04-19:
收到，多谢

wstong999, 2022-04-19:
[image]
我们源码的接口有5个参数，如果你们自定义接口需要传了6个参数，那对应的接口需要修改

rualdeng, 2022-04-19:
> [image] 我们源码的接口有5个参数，如果你们自定义接口需要传了6个参数，那对应的接口需要修改

我用的是oss_upload_file这个函数，不是你们原码的 oss_upload_local_file这个函数。
[image]
[image]
我把你们原码oss_upload_local_file这个函数注释了，用了oss_upload_file这个函数。

rualdeng, 2022-04-19:
这个oss_upload_file函数是你们OSS里本身就有的

rualdeng, 2022-04-20:
不知道我描述清楚没，因为业务需求我把原码中的oss_upload_local_file函数换成OSS中原有的oss_upload_file函数，然后出现内存泄露，每次上传20几K文件消耗700字节。所以麻烦查一下oss_upload_file这个函数的内存泄露问题。谢谢

rualdeng, 2022-04-20:
> [image] 我们源码的接口有5个参数，如果你们自定义接口需要传了6个参数，那对应的接口需要修改

不知道我描述清楚没，因为业务需求我把原码中的oss_upload_local_file函数换成OSS中原有的oss_upload_file函数，然后出现内存泄露，每次上传20几K文件消耗700字节。所以麻烦查一下oss_upload_file这个函数的内存泄露问题。谢谢

wstong999, 2022-04-25:
这个不是内存泄漏，GC过一会自动会回收释放，看log，一开始2849620-》2849396-》2848708-》2848472-》2848012-》2849884

[free-defaut] **2849620**
[free-8bit-internal] 45408
[free-8bit-spiram] 2804300
>>> 
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> 
>>> uos.heap()
[free-defaut] 2849396
[free-8bit-internal] 45184
[free-8bit-spiram] 2804300
>>> uos.heap()
[free-defaut] **2849396**
[free-8bit-internal] 45184
[free-8bit-spiram] 2804300
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> uo␛[Ks.heap()
[free-defaut] **2848708**
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> uos.heap()
[free-defaut] 2848708
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> uos.heap()
[free-defaut] 2848708
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> uos.heap()
[free-defaut] 2848708
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> uos.heap()
[free-defaut] 2848708
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> uos.heap()
[free-defaut] 2848708
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> uos.heap()
[free-defaut] 2848708
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> content = "test data"
>>> file = oss.uploadContent(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, content, "oss/test.txt")
>>> 
>>> uos.heap()
[free-defaut] 2848708
[free-8bit-internal] 44496
[free-8bit-spiram] 2804300
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> 
>>> uos.heap()
[free-defaut] **2848472**
[free-8bit-internal] 44260
[free-8bit-spiram] 2804300
>>> 
>>> 
>>> 
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> 
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, 'data/pyamp/test.py',OBJECT_NAME)
>>> uos.heap()
[free-defaut] **2848012**
[free-8bit-internal] 43800
[free-8bit-spiram] 2804300
>>> 
>>> 
>>> 
>>> uos.heap()
[free-defaut] **2849884**
[free-8bit-internal] 45672
[free-8bit-spiram] 2804300
>>>

rualdeng, 2022-04-25:
> 这个不是内存泄漏，GC过一会自动会回收释放，看log，一开始2849620-》2849396-》2848708-》2848472-》2848012-》2849884
>
> [free-defaut] **2849620** [free-8bit-internal] 45408 [free-8bit-spiram] 2804300

非常感谢，之前也注意到会回收，但是上传完后等待一定时间才回收。我的业务逻辑是需要频繁上传，所以怕还没有回收或回收没有耗得快，把耗尽资源了。

rualdeng, 2022-04-25:
> 这个不是内存泄漏，GC过一会自动会回收释放，看log，一开始2849620-》2849396-》2848708-》2848472-》2848012-》2849884
>
> [free-defaut] **2849620** [free-8bit-internal] 45408 [free-8bit-spiram] 2804300

非常感谢，之前也注意到会回收，但是上传完后等待一定时间才回收。貌似我用gc主动回收没用。我的业务逻辑是需要频繁上传，所以怕还没有回收或回收没有耗得快，把耗尽资源了。

closed 2022-05-06 by skylarCai