alibaba/AliOS-Things #1493: [Bug Report]: m5用aliyunIot Device 上传 报BUG

rualdeng, 2022-04-19:
### Contact Details

rualdeng@163.com

### What happened?

A bug happened!
运行环境:
固件版本:最新版本

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

### Relevant log output

```shell
[  12.690]<E>AIOT_UPLOAD open File:1.amr failed 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x402fc150  PS      : 0x00060930  A0      : 0x802fc10d  A1      : 0x3fff0b20  
A2      : 0x3fff0fdc  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ff96458  A7      : 0x00000000  A8      : 0x802fc150  A9      : 0x3fff0b00  
A10     : 0x00000000  A11     : 0x3fff0b5c  A12     : 0x3ffbe51c  A13     : 0x00000000  
A14     : 0x3ffd933c  A15     : 0x3fff0fdc  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000064  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x402fc14d:0x3fff0b20 0x402fc10a:0x3fff0b90 0x4010e4a3:0x3fff0bb0 0x4024dc58:0x3fff0be0 0x4024e0ce:0x3fff0c10 0x40245f32:0x3fff0c80 0x40248229:0x3fff0e30 0x4010ddea:0x3fff0ed0

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

代码为：
def uploadFile():
    global dev
    begin = time.time()
    ret = dev.uploadFile("1.amr", "data/1.amr")
    if ret != None :
        print('上传成功')
    else :
        print('上传失败')
    end = time.time()
    print('the time in upload() ' ,end - begin)

同时dev.uploadFile  按照文案ret = aliyunIoT_device.uploadFile("aa.jpg", "data/pyamp/1.jpg", None)写，报只有三个参数，确给了4个参数的错。所以把None去掉了。
文件大小为14K
```

skylarCai, 2022-04-19:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

senwangal, 2022-04-19:
目前因为M5 的内存限制，文件大小只能支持10K以下的，超出会有内存不足的问题，这个问题我们还在优化，当前建议用10K以下的图片测试

closed 2022-04-25 by skylarCai