alibaba/AliOS-Things #1495: [Bug Report]: m5报内存泄露

rualdeng, 2022-04-22:
### Contact Details

rualdeng@163.com

### What happened?

A bug happened!
运行环境:m5
固件版本:最新版本
设备连上WIFI和上云后，订阅自定义消息。接收云上的命令执行_thread.start_new_thread(record_start,())，出现内存泄露。单独_thread.start_new_thread 录音，不加OSS不会内存泄露，加上上传直接重启。麻烦帮忙解决一下。谢谢

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

### Relevant log output

```shell
CORRUPT HEAP: multi_heap.c:438 detected at 0x3fff10e0

abort() was called at PC 0x40097e3f on core 0

Backtrace:0x40093dd1:0x3fff04b0 0x400943bd:0x3fff04d0 0x400989d1:0x3fff04f0 0x40097e3f:0x3fff0560 0x400981fc:0x3fff0580 0x400986c4:0x3fff05b0 0x40085e7e:0x3fff05d0 0x40085ea2:0x3fff0600 0x400989d9:0x3fff0620 0x402d15f9:0x3fff0640 0x402dda07:0x3fff0660 0x40299d26:0x3fff0680 0x40299d63:0x3fff06b0 0x4029b142:0x3fff06d0 0x402a6d7e:0x3fff19d0 0x4029481e:0x3fff1a20 0x40291980:0x3fff1a60 0x40295f88:0x3fff1b60 0x4029608e:0x3fff1c00 0x40292092:0x3fff1cb0 0x402923d7:0x3fff1da0 0x4028d611:0x3fff1ed0 0x4029516f:0x3fff1ef0 0x40112d73:0x3fff2260 0x400ef095:0x3fff22a0 0x400f7969:0x3fff22d0 0x400f7aba:0x3fff22f0 0x400fa925:0x3fff2310 0x400ef120:0x3fff23b0 0x400f7969:0x3fff23e0 0x400fa876:0x3fff2400 0x400ef120:0x3fff24a0 0x400f7969:0x3fff2510 0x400fa876:0x3fff2530 0x400ef120:0x3fff25d0 0x400f7969:0x3fff2630 0x400ea433:0x3fff2650 0x4010b569:0x3fff26f0

ELF file SHA256: 0000000000000000

Rebooting...

def upload():
    #global filename
    ACCESS_KEY = ''
    ACCESS_SECRET = ''
        # OSS对象存储配置
    OSS_ENDPOINT = ""
   
    OSS_BUCKET = ""
        # test.jpg存放在“/”目录下
    begin = time.time()
    file = oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, '1.amr')
    end = time.time()
    print('the time in upload() ' ,end - begin)

    return file

def record_start():
    global rec
    if rec == None:
        print("this is in rec==None")
        rec=Recorder()
    rec.start(uri='file://1.amr', format=rec.AMR)
    time.sleep(5)
    rec.stop()
    print("the record is stop")
    time.sleep(1)
    fileurl = upload()

在订阅中执行_thread.start_new_thread(record_start,()) 出现内存泄露
```

skylarCai, 2022-04-22:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yanxiaoyong-ch, 2022-04-22:
fileurl = upload()加上后会报如下的错误吗：

ORRUPT HEAP: multi_heap.c:438 detected at 0x3fff10e0

abort() was called at PC 0x40097e3f on core 0

Backtrace:0x40093dd1:0x3fff04b0 0x400943bd:0x3fff04d0 0x400989d1:0x3fff04f0 0x40097e3f:0x3fff0560 0x400981fc:0x3fff0580 0x400986c4:0x3fff05b0 0x40085e7e:0x3fff05d0 0x40085ea2:0x3fff0600 0x400989d9:0x3fff0620 0x402d15f9:0x3fff0640 0x402dda07:0x3fff0660 0x40299d26:0x3fff0680 0x40299d63:0x3fff06b0 0x4029b142:0x3fff06d0 0x402a6d7e:0x3fff19d0 0x4029481e:0x3fff1a20 0x40291980:0x3fff1a60 0x40295f88:0x3fff1b60 0x4029608e:0x3fff1c00 0x40292092:0x3fff1cb0 0x402923d7:0x3fff1da0 0x4028d611:0x3fff1ed0 0x4029516f:0x3fff1ef0 0x40112d73:0x3fff2260 0x400ef095:0x3fff22a0 0x400f7969:0x3fff22d0 0x400f7aba:0x3fff22f0 0x400fa925:0x3fff2310 0x400ef120:0x3fff23b0 0x400f7969:0x3fff23e0 0x400fa876:0x3fff2400 0x400ef120:0x3fff24a0 0x400f7969:0x3fff2510 0x400fa876:0x3fff2530 0x400ef120:0x3fff25d0 0x400f7969:0x3fff2630 0x400ea433:0x3fff2650 0x4010b569:0x3fff26f0

ELF file SHA256: 0000000000000000

Rebooting...

rualdeng, 2022-04-22:
> fileurl = upload()加上后会报如下的错误吗：
>
> ORRUPT HEAP: multi_heap.c:438 detected at 0x3fff10e0
>
> abort() was called at PC 0x40097e3f on core 0
>
> Backtrace:0x40093dd1:0x3fff04b0 0x400943bd:0x3fff04d0 [...]

是的

rualdeng, 2022-04-23:
> fileurl = upload()加上后会报如下的错误吗：
>
> ORRUPT HEAP: multi_heap.c:438 detected at 0x3fff10e0
>
> abort() was called at PC 0x40097e3f on core 0
>
> Backtrace:0x40093dd1:0x3fff04b0 0x400943bd:0x3fff04d0 [...]

进展如何，该如何处理这问题？

yanxiaoyong-ch, 2022-04-24:
脚本、board.json、lib等文件系统里面的可复现环境麻烦发出来，我们同步复现

rualdeng, 2022-04-25:
> 脚本、board.json、lib等文件系统里面的可复现环境麻烦发出来，我们同步复现

加下钉钉吧，直接对接效率高一些。这样效率比较低。钉钉号：rualxiao

rualdeng, 2022-04-25:
> ```shell
> def upload():
>     #global filename
>     ACCESS_KEY = ''
>     ACCESS_SECRET = ''
>         # OSS对象存储配置
>     OSS_ENDPOINT = ""
>    
>     OSS_BUCKET = ""
>         # test.jpg存放在“/”目录下
>     begin = time.time()
>     file = oss.uploadFile(ACCESS_KEY, ACCESS_SECRET, OSS_ENDPOINT, OSS_BUCKET, '1.amr')
>     end = time.time()
>     print('the time in upload() ' ,end - begin)
> 
>     return file
> 
> def record_start():
>     global rec
>     if rec == None:
>         print("this is in rec==None")
>         rec=Recorder()
>     rec.start(uri='file://1.amr', format=rec.AMR)
>     time.sleep(5)
>     rec.stop()
>     print("the record is stop")
>     time.sleep(1)
>     fileurl = upload()
> ```

因为有一些是特定的订阅和一些逻辑，不及时沟通，发出来也没办法复现。麻烦工程师能加个钉钉直接对接沟通，谢谢！

rualdeng, 2022-04-25:
> 脚本、board.json、lib等文件系统里面的可复现环境麻烦发出来，我们同步复现

因为有一些是特定的订阅和一些逻辑，不及时沟通，发出来也没办法复现。麻烦工程师能加个钉钉直接对接沟通，谢谢！