alibaba/AliOS-Things #1502: [Bug Report]: esp32上推送helloworld例程失败

zeng2008, 2022-04-26:
### Contact Details

2768724163@qq.com

### What happened?

A bug happened!
运行环境:Windows10
固件版本:HaaSPython-ESP32-v2.0.0

### Version

rel_3.3.0 (Main)

### What soultions are you seeing the problem on?

helloworld_demo

### Relevant log output

```shell
Windows PowerShell
版权所有 (C) Microsoft Corporation。保留所有权利。

尝试新的跨平台 PowerShell https://aka.ms/pscore6

PS D:\1电科实训\projects\Haas projects\HaaSPython> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\86178\.vscode\extensions\haas.haas-studio-2.0.0\asserts\mpy_tools\mpytool.py -f d:\1 
电科实训\projects\Haas projects\HaaSPython\solutions\test_demo -t c:\Users\86178\.vscode\extensions\haas.haas-studio-2.0.0\asserts\mpy_tools  
COM3
[*] Filename is  d:\1电科实训\projects\Haas
Traceback (most recent call last):
  File "c:\Users\86178\.vscode\extensions\haas.haas-studio-2.0.0\asserts\mpy_tools\mpytool.py", line 105, in <module>
    sys.path.append(tool)
NameError: name 'tool' is not defined
PS D:\1电科实训\projects\Haas projects\HaaSPython>
```

skylarCai, 2022-04-26:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-04-26:
我们先试下

zetamind, 2022-04-26:
这个 tool 就是 mpytool 的路径。 从你的 log 上看，这个 tool 的参数是往下传递了的，c:\Users\86178\.vscode\extensions\haas.haas-studio-2.0.0\asserts\mpy_tools
这个就是 tool 的定义，但是在 python 执行的时候，没有获取到这个值。 请问下你是使用管理员权限打开的 vscode 吗 ？ 对 C 盘的文件是否有足够的操作权限？

closed 2022-05-18 by skylarCai