alibaba/AliOS-Things #1513: [Bug Report][AutoTest]: 偶现aliyuniot device 获取ntp时间失败

PiaoLingAl, 2022-04-27:
### Contact Details

guoliang.wgl@alibaba-inc.com

### What happened?

A bug happened!
运行环境: ubuntu
固件版本: 最新固件

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_demo

### Relevant log output

```shell
MicroPython v1.17-dirty on 2022-04-25T15:43:47; ESP32 module with ESP32
Type "help()" for more information.
AosAutoTest:SetupEnv: succeed
Task Name	Status	Prio	HWM	Taskid#
mp_task        	R	1	13136	7	0

IDLE0          	R	0	1132	4	0

tiT            	B	18	2268	9	-1

queue_handler  	B	3	7712	6	0

Tmr Svc        	B	1	1704	5	0

sys_evt        	B	20	1848	10	0

mdns           	B	1	2580	12	0

wifi           	B	23	4644	11	0

esp_timer      	B	22	3460	1	0

ipc0           	B	24	628	2	0

[free-defaut] 121684

[free-8bit-internal] 62240

Task: Pre-Scheduler allocs -> CAP_INTERNAL_8BIT: 5500 CAP_SPIRAM_8BIT: 0
Task: esp_timer -> CAP_INTERNAL_8BIT: 1800 CAP_SPIRAM_8BIT: 0
Task: ipc0 -> CAP_INTERNAL_8BIT: 4984 CAP_SPIRAM_8BIT: 0
Task: �t�? -> CAP_INTERNAL_8BIT: 28468 CAP_SPIRAM_8BIT: 0
Task: mp_task -> CAP_INTERNAL_8BIT: 117632 CAP_SPIRAM_8BIT: 0
Task: tiT -> CAP_INTERNAL_8BIT: 380 CAP_SPIRAM_8BIT: 0
Task: sys_evt -> CAP_INTERNAL_8BIT: 5864 CAP_SPIRAM_8BIT: 0
Task: queue_handler -> CAP_INTERNAL_8BIT: 304 CAP_SPIRAM_8BIT: 0
Task: wifi -> CAP_INTERNAL_8BIT: 11712 CAP_SPIRAM_8BIT: 0
Task: mdns -> CAP_INTERNAL_8BIT: 244 CAP_SPIRAM_8BIT: 0

[internal]

Heap summary for capabilities 0x00000800:
  At 0x3ff80064 len 8092 free 0 allocated 7484 min_free 0
    largest_free_block 0 alloc_blocks 28 free_blocks 1 total_blocks 29
  At 0x3ffaff10 len 240 free 0 allocated 88 min_free 0
    largest_free_block 0 alloc_blocks 6 free_blocks 0 total_blocks 6
  At 0x3ffb6388 len 7288 free 0 allocated 5772 min_free 0
    largest_free_block 0 alloc_blocks 73 free_blocks 2 total_blocks 75
  At 0x3ffb9a20 len 16648 free 4 allocated 15704 min_free 0
    largest_free_block 0 alloc_blocks 44 free_blocks 3 total_blocks 47
  At 0x3ffcbf88 len 82040 free 30380 allocated 49764 min_free 26920
    largest_free_block 30124 alloc_blocks 91 free_blocks 7 total_blocks 98
  At 0x3ffe0440 len 129984 free 31608 allocated 98304 min_free 31608
    largest_free_block 31608 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x40078000 len 32768 free 32716 allocated 0 min_free 32716
    largest_free_block 32716 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x40099764 len 26780 free 26728 allocated 0 min_free 26728
    largest_free_block 26728 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 121684 allocated 176888 min_free 117972 largest_free_block 32716
HaaSPython-ESP32-v2.1.0-2022-04-25T15:43:47

AosAutoTest:----------------------------------mod:IotDevice start-----------------------------------

AosAutoTest:ModStart:start modtest!
AosAutoTest:CaseStart: test_init (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_attr (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_on (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_register (IotDevice) ...
establish mbedtls connection with server(host='iot-auth.cn-shanghai.aliyuncs.com', port=[443])
tcp_connect: can only connect from state CLOSED
success to establish tcp, fd=54
success to establish mbedtls connection, fd = 54(cost 0 bytes in total, max used 0 bytes)
c1f96538e4c615e5174e6f6c5f44b806
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_on_register (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_connect (IotDevice) ...
establish tcp connection with server(host='a1uTFk4xjko.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_on_connect (IotDevice) ...
tcp_connect: can only connect from state CLOSED
success to establish tcp, fd=54
lk connected
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_getdevicehandle (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_getntptime (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_on_getntptime (IotDevice) ...
AosAutoTest:CaseResult: FAIL 
AosAutoTest:CaseStart: test_post_props (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_on_props (IotDevice) ...
AosAutoTest:Condition:1:
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_post_event (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_on_service (IotDevice) ...
AosAutoTest:Condition:2:
2022-4-25 15:47:27:592
timestamp: 1650872847
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_subcribe_publish_unsubcribe (IotDevice) ...
AosAutoTest:CaseResult: ok 
AosAutoTest:CaseStart: test_end (IotDevice) ...
lk disconnected
AosAutoTest:CaseResult: ok 
AosAutoTest:Run 16 tests

AosAutoTest:ModResult: FAILED (failures=1, errors=0)

AosAutoTest:******************************************************************

['AosAutoTest:', 'AosAutoTest:Traceback (most recent call last):', 'AosAutoTest:  File "unittest.py", line 222, in run_class', 'AosAutoTest:  File "<stdin>", line 342, in test_on_getntptime', 'AosAutoTest:  File "unittest.py", line 113, in assertTrue', 'AosAutoTest:AssertionError: Expected False to be True', 'AosAutoTest:']
AosAutoTest:
AosAutoTest:Traceback (most recent call last):
AosAutoTest:  File "unittest.py", line 222, in run_class
AosAutoTest:  File "<stdin>", line 342, in test_on_getntptime
AosAutoTest:  File "unittest.py", line 113, in assertTrue
AosAutoTest:AssertionError: Expected False to be True
AosAutoTest:
AosAutoTest:******************************************************************
```

skylarCai, 2022-04-27:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-05-17:
请问这个问题解决了吗？
如果已经解决，请关闭

victorwangstone, 2022-05-17:
偶现的错误，和网络延时相关，无法复现，可先关闭观察。已经提出修改test_on_getntptime 测试用例的方案：将timeout值增加。 如果错误发生的频率过多，可通过实施测试修改方案，降低错误出现的概率。

closed 2022-05-18 by skylarCai