alibaba/AliOS-Things #1569: 开源之夏2022开啦！ HaaS社区精选赛题正式上线！ 助力国产开源社区建设，赢万元项目奖金！

ElliottSeer, 2022-05-06:
# 活动介绍

开源之夏是由“开源软件供应链点亮计划”发起并长期支持的一项暑期开源活动，旨在鼓励在校学生积极参与开源软件的开发维护，促进优秀开源软件社区的蓬勃发展，培养和发掘更多优秀的开发者。

活动联合国内外各大开源社区，针对重要开源软件的开发与维护提供项目任务，并面向全球高校学生开放报名。

点击这里前往 [活动官网](https://summer-ospp.ac.cn/)

# 社区概况

AliOS Things 发布于2017年杭州云栖大会， 是 面向 IoT 领域的、高可伸缩的物联网操作系统。

AliOS Things 致力于搭建云端一体化 IoT 基础设施，具备极致性能、极简开发、云端一体、丰富组件、安全防护等关键能力。AliOS Things 支持多种多样的设备连接到阿里云物联网平台，可广泛应用在智能家居、智慧城市、工业，新出行等领域。

[仓库地址](https://github.com/alibaba/AliOS-Things)

# 社区项目

## Arm Virtual Hardware上实现AliOS Things移植

Arm Virtual Hardware是基于云端的虚拟开发平台，为应用程序开发人员提供虚拟的基于 Arm 的 SoC，同时虚拟了内存和外围设备，为敏捷软件开发（包括CI/CD、DevOps和MLOps工作流）提供快速便利的开发和测试环境。

项目难度：进阶

项目奖金：¥12000

赛题地址：https://summer-ospp.ac.cn/#/org/prodetail/226600075

##  基于开源MQTT自建简易物联网平台实现HaaS Python接入

本项目的目标是通过开源的MQTT物联网平台框架（如Thingsboard，EMQ等），配合云服务器，手动打造一台支持HaaS Python接入的简易物联网云平台。

项目难度：进阶

项目奖金：¥12000

赛题地址： https://summer-ospp.ac.cn/#/org/prodetail/226600077

# 申请攻略

## 申请资格

- 本活动面向**年满 18 周岁**的**在校学生**。
- 暑期即将毕业的学生，只要在申请时学生证处在有效期内，就可以提交申请。
- 海外学生可提供录取通知书/学生卡/在读证明证明学生身份。

## 申请建议

- 仔细阅读(学生指南)[https://summer-ospp.ac.cn/help/student]
- 多与导师沟通：对开源项目有一个全面的认识；细化方案，提升可行性；展现良好的沟通能力，有助于提升中选概率。
- 每周至少 30 个小时能够投入在项目中。
- 对活动充分了解，至少要阅读学生指南、热点问题、活动日程等内容。
- 仔细查看社区的文档，认真熟悉社区的参与方式、沟通渠道、社区软件设计文档、社区开源仓库代码等。

## 社区支持

HaaS官网 提供开发案例及丰富资源
https://haas.iot.aliyun.com/

参赛过程中需要的任何问题，可以在该Issue下反馈，或通过以下途径及时交流

赛事支持钉钉群：
[image]

HaaS开发者群及微信公众号：
[image]

skylarCai, 2022-05-25:
[image]

skylarCai, 2022-05-26:
[image]