alibaba/AliOS-Things #1583: [Bug Report]: haas506 调用modbus.deinit 自动重启

coonyshu, 2022-05-08:
### Contact Details

19012315@qq.com

### What happened?

A bug happened!
运行环境: macOS 12.3.1, vscode 1.64 haas-studio 2.1.0
固件版本: 2.0.1

在485没有外接设备的情况下，一直无限自动重启，1.67固件版本都是正常的

[1651977724.931]<E>BOARD_MGR board_get_items failed, name_id = modbus

[1651977724.935]<E>modbus parser_from_board_configuration_file: py_board_attach_item failed ret = amp shakehand begin...
POWERONREASON:0x0208,parse:unknown,charge,panic.

borad:

"modbus":{
      "type":"MODBUS",
      "mode":0,
      "port":2,
      "baudrate":9600,
      "timeout":1000,
      "parity":0
    }

代码如下：
modbus.init('modbus')
    #utime.sleep(1)
    print('Reading Pressure Data')

    # 主机读寄存器： 设备ID是1，寄存器起始地址为0，连续读入2个寄存器数据， 响应超时为持续等待
    pressuredata = bytearray(4)
    ret1 = modbus.readHoldingRegisters(1, 0, 2, pressuredata, 1000)

    # 将十六进制读数转化为十进制压力
    global pressure, pressure_data, pressure_data_str
    pressure = (ustruct.unpack('!f',pressuredata[3:]+pressuredata[2:3]+pressuredata[1:2]+pressuredata[0:1]))[0]
    pressure = int(pressure * 100)
    pressure_data = bytearray(4)
    gl_val.pressure_data = ustruct.pack('!h',pressure)
    print('pressure:', pressure)

    # 读取温压传感器中的温度
    utime.sleep(1)
    print('Reading Temperature Data')
    tempdata = bytearray(4)
    ret2 = modbus.readHoldingRegisters(1, 2, 2, tempdata, 1000)

    # 将十六进制读数转化为十进制温度
    temperature = (ustruct.unpack('!f',tempdata[3:]+tempdata[2:3]+tempdata[1:2]+tempdata[0:1]))[0]  
    print('temperature:', temperature)
    temperature = int(temperature * 100)
    gl_val.temperature_data= ustruct.pack('!h',temperature)
    gl_val.templist.append(gl_val.temperature_data)
    gl_val.templist1.append(gl_val.temperature_data)
    gl_val.presslist.append(gl_val.pressure_data)
    gl_val.presslist1.append(gl_val.pressure_data)

    # 读取温湿传感器中的温度
    utime.sleep_ms(10)
    print('Reading Temperature Data from temp sensor')
    temphum = bytearray(4)
    # 主机读寄存器： 设备ID是2，寄存器起始地址为0，连续读入2个寄存器数据， 响应超时为持续等待
    ret1 = modbus.readHoldingRegisters(2, 0, 2, temphum, 1000)
    temp =  ustruct.unpack('h',temphum[0:2])[0]
    print('the temperatur is ', temp)
    gl_val.temp_data = ustruct.pack('!h',temp)
    #print('temphum',gl_val.temp_data)
    gl_val.temp_sensor_list.append(gl_val.temp_data)
    gl_val.temp_sensor_list1.append(gl_val.temp_data)

    modbus.deinit()

### Version

master (Default)

skylarCai, 2022-05-08:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zy20220224, 2022-05-10:
[image]
看这行log，可以发现board.json没有配置有问题。

因此，modbus.init('modbus') 执行失败了。 这里需要判断返回状态的。

YiluMao, 2022-05-17:
@coonyshu