alibaba/AliOS-Things #1591: [Feature Request]: 请问haaseduk1有使用气压传感器和运动传感器的案例吗

krisaaaaa, 2022-05-09:
### What do you want?

Haas edu k1开发板载有:加速度传感器 | MPU6050、气压传感器 | SPL06，
请问有没有用到这两个传感器的相关案例呢？

### Version

master (Default)

skylarCai, 2022-05-09:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-05-09:
> ### Contact Details
> _No response_
> 
> ### What do you want?
> Haas edu k1开发板载有:加速度传感器 | MPU6050、气压传感器 | SPL06， 请问有没有用到这两个传感器的相关案例呢？
> 
> ### Version
> master (Default)

目前咱没有使用这两个传感器的趣味案例，但这两个传感器的驱动代码都是有开源出来的。
HaaS EDU K1有两个不同的硬件版本，硬件版本的判定及其搭载的传感器及传感器的驱动代码连接可以参考：
https://haas.iot.aliyun.com/solution/detail/hardware?versionId=800C7EC389F878E500000001&dataId=800C7EC389F878E5
[image]

krisaaaaa, 2022-05-09:
好的，谢谢。

ethanlcz, 2022-05-09:
> 好的，谢谢。

不客气，使用上有任何问题欢迎联系我们。

closed 2022-05-09 by skylarCai

krisaaaaa, 2022-05-10:
@skylarCai 
你好，此目录里的文件我下载不到。
[image]
[image]

ethanlcz, 2022-05-10:
> @skylarCai 你好，此目录里的文件我下载不到。 [image] [image]

请从此链接下载。
https://github.com/alibaba/AliOS-Things/tree/master/haas_lib_bundles/python/libraries/haaseduk1/code

krisaaaaa, 2022-05-10:
> > @skylarCai 你好，此目录里的文件我下载不到。 [image] [image]
> 
> 请从此链接下载。 https://github.com/alibaba/AliOS-Things/tree/master/haas_lib_bundles/python/libraries/haaseduk1/code

OK，谢谢。