alibaba/AliOS-Things #1593: [Feature Request]: Hass edu k1怎么同时运行两个案列

krisaaaaa, 2022-05-09:
### What do you want?

我想同时运行几个案例，比如同时实现温湿度上云和光照强度上云，可以像这样直接将两个案例的文件放在solutions下吗（humiture_py1是温湿度上云案例，light是光强度上云案例）

[image]

我这样部署运行了一下，但是好像只运行了一个。请问是需要将两个案例整合在一起才行吗，或者要怎么操作呢

### Version

master (Default)

skylarCai, 2022-05-09:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-05-09:
> ### Contact Details
> _No response_
> 
> ### What do you want?
> 我想同时运行几个案例，比如同时实现温湿度上云和光照强度上云，可以像这样直接将两个案例的文件放在solutions下吗（humiture_py1是温湿度上云案例，light是光强度上云案例）
> 
> [image]
> 
> 我这样部署运行了一下，但是好像只运行了一个。请问是需要将两个案例整合在一起才行吗，或者要怎么操作呢
> 
> ### Version
> master (Default)

Python引擎值会解析main.py这个文件，所以如果想在一个solution中用两个传感器的功能，需要从这两个main.py中找到对自己有帮助的代码，合并成一个main.py，再推送到设备中去。

ethanlcz, 2022-05-16:
hi  krisaaaaa,
请问下这个问题解决了没？